CloverHackyColor/CloverBootloader #657: Some of Borad Can't boot with Beta version

dateno1, 2024-01-06:
[GA-B365M-AE EFI 240106.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13848630/GA-B365M-AE.EFI.240106.zip)
EFI (Using Version 5155) Attatched (It not updated/Serial Removed)

I was success to boot&Recovery BCM94360 with beta version on two board but I was fail to boot with GA-B365M Aorus Elite

Test with 'beta-bd46eeffe' + Sonoma Lastest (14.2.1)

If i Replace EFI (or EFI&Drivers) can't boot with this board

No error (just black screen&Stuck)

SergeySlice, 2024-01-08:
VirtualSMC is not compatible with Clover. Use FakeSMC instead. And correspondent plugins.

SergeySlice, 2024-01-08:
You also should provide preboot.log (by F2) and ACPI tables (by F4) to show full picture of the issue.

dateno1, 2024-01-08:
> VirtualSMC is not compatible with Clover. Use FakeSMC instead. And correspondent plugins.

I don't think so

Clover 5155 has no problem with VirtualSMC and Other computers used Clover with VirtualSMC

I was used clover with VirtualSMC from Mojave to Sonoma on multiple computers (Some of computer Sensors not work with FakeSMC)

I think Old clover (Pre OC/~5122) is better with FakeSMC but lastest Clover using OC (It recommended using with VirtualSMC)

Atattched EFI work well (exclude BlockSkywalk quirk)

> You also should provide preboot.log (by F2) and ACPI tables (by F4) to show full picture of the issue.

5115 : [5155.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13859308/5155.zip)
bd46eeffe : [bd46eeffe.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13859310/bd46eeffe.zip)

No picture can be shot (black screen)

jief666, 2024-01-09:
preboot.log is not enough. There is none of the OC messages. You must set the "debug" key to true to generate a file name something like 2024-01-09_15-06_CloverX64.efi.log.
Do you have a way to boot this motherboard with an older Clover or an OpenCore config ?

dateno1, 2024-01-09:
> preboot.log is not enough. There is none of the OC messages. You must set the "debug" key to true to generate a file name something like 2024-01-09_15-06_CloverX64.efi.log. Do you have a way to boot this motherboard with an older Clover or an OpenCore config ?

Which one do you need? (5155? beta?)

All EFI uploaded (I just updated clover 5155 -> bd46eef)

You can check config file from EFI File

Question : How can i get debug log? (I tried Search 'debug' on config file but nothing found)

jief666, 2024-01-10:
Let's go step by step : Do you have a way to boot this motherboard with Clover or OpenCore ?

dateno1, 2024-01-10:
> [GA-B365M-AE EFI 240106.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13848630/GA-B365M-AE.EFI.240106.zip) EFI Attatched (It not updated/Serial Removed)
> 

You can check all files&settings from here

This board work well with Clover 5155 (exclude BlockSkywalk)

I was tried update Clover (Replace BOOTX64.efi, CLOVERX64.efi, Tools, Drivers) for fix broadcom wireless

Now I can't boot sonoma on this board

jief666, 2024-01-10:
If the board works well with 5155, and now it's not anymore because of some update, you may try the forum. I'm sure some user will inspect your zip.
I also suggest you revert back to a clover folder that works and then update files one by one.
Sorry, but I can't program Clover AND fix configuration/update problems.

dateno1, 2024-01-10:
> If the board works well with 5155, and now it's not anymore because of some update, you may try the forum. I'm sure some user will inspect your zip. I also suggest you revert back to a clover folder [...]

Already tested

Replace Cloverx64.efi = No boot

SergeySlice, 2024-01-25:
> > VirtualSMC is not compatible with Clover. Use FakeSMC instead. And correspondent plugins.
> 
> I don't think so
> 
> Clover 5155 has no problem with VirtualSMC and Other computers used Clover with VirtualSMC
> 
> I was used clover with VirtualSMC from Mojave to Sonoma on multiple computers (Some of computer Sensors not work with FakeSMC)
> 
> I think Old clover (Pre OC/~5122) is better with FakeSMC but lastest Clover using OC (It recommended using with VirtualSMC)
> 
> Atattched EFI work well (exclude BlockSkywalk quirk)
> 
> > You also should provide preboot.log (by F2) and ACPI tables (by F4) to show full picture of the issue.
> 
> 5115 : [5155.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13859308/5155.zip) [bd46eef](https://github.com/CloverHackyColor/CloverBootloader/commit/bd46eeffe91c474abfca0b3155ba0a1e679c73f2) : [bd46eeffe.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13859310/bd46eeffe.zip)
> 
> No picture can be shot (black screen)

OK, but new Clover contains new OpenCore which blocks using VirtualSMC.

dateno1, 2024-01-25:
> OK, but new Clover contains new OpenCore which blocks using VirtualSMC.

I already success to boot Sonoma with Clover + VirtualSMC on another two boards

jief666, 2024-01-26:
@dateno1 You said you successfully boot 2 boards. Is this issue still open ?

You sent a zip, but does it contains successful boot (with debug log) ? The unsuccessful boot ?
I'm sorry but yon can't just send a zip and hoping that we'll do all the work guessing what is what.

So first, send me a log with a successful boot with an old clover, and then a log with the unsuccessful boot with the latest Clover from the repo. (If you can't compile yourself, I can send you a Clover efi file).
Thanks.

dateno1, 2024-01-26:
> @dateno1 You said you successfully boot 2 boards. Is this issue still open ?
>
> You sent a zip, but does it contains successful boot (with debug log) ? The unsuccessful boot ? I'm sorry but yon can't [...]

I was more boards (not only 2)

Problem is clover work with some of boards but not work with some of boards

> Which one do you need? (5155? beta?)
>
> All EFI uploaded (I just updated clover 5155 -> [...]

How can i get debug log?

I don't know how can i get that

I wasn't compiled it myself (Downloaded from this github's action)