CloverHackyColor/CloverBootloader #659: Incorrect orientation after replacing the screen with a better version.

AllevatoreAnonimo, 2024-01-08:
After replacing the stock LCD screen with its improved version (a DeckHD to be precise) it is rotated as in the attached image.
[image]

SergeySlice, 2024-01-08:
Huh! What is the device?

jief666, 2024-01-09:
wow, clover ends up in unexpected places ! :-)
Once booted, do you have the correct orientation ? Can you get a log from that ?

AllevatoreAnonimo, 2024-01-10:
> Huh! What is the device?

I can guarantee that under the hood it still has the heart of a Steam Deck.
I think the problem came up when it switched to 1900x1200 resolution.

> wow, clover ends up in unexpected places ! :-)
> Once booted, do you have the correct orientation ? Can you get a log from that ?

Yes, once started the screen is in the right orientation.
What logs do you need?

[image]

jief666, 2024-01-10:
There is a key that you can add to enable debug log. Then send me that. If you're not sure what key is that, ask on the forum.

AllevatoreAnonimo, 2024-01-10:
I found within Clover-Toolbox a dedicated function for those with the DeckHD mod that forces the correct resolution to 1900x1200 instead of 1200x1900.
Closing the issue.
[image]
[image]

closed 2024-01-10 by AllevatoreAnonimo

jief666, 2024-01-10:
Can you tell what does that tool do ? Setting a parameter in config.plist ?

jief666, 2024-01-10:
I found the script source, I think.
I can see inside that it add this key to config.plist :
<key>ScreenResolution<key><string>1920x1200<string>
I've no idea why that would rotate the screen 90 degrees !

AllevatoreAnonimo, 2024-01-10:
> I found the script source, I think.
> I can see inside that it add this key to config.plist :
> `<key>ScreenResolution<key><string>1920x1200<string>`
> I've no idea why that would rotate the screen 90 degrees !

I don't know, I saw that it is also possible to modify through this tool the timeout for the automatic choice and even the type of theme to be used. 
It's probably going to tweak some parameters in that file.