CloverHackyColor/CloverBootloader #660: Incorrect display of the order of lines with memory slots in the profiler.

ybronst, 2024-01-09:
The order of the memory slot lines in the profiler does not match the actual one.
Instead of:
Channel1-Dimm1 - empty
Channel1-Dimm0 - 32 GB
Channel0-Dimm1 - empty
Channel0-Dimm0 - 32 GB
displayed:
Channel1-Dimm1 - empty
Channel1-Dimm0 - empty
Channel0-Dimm1 - 32 GB
Channel0-Dimm0 - 32 GB
Using the bdmesg command, everything is determined correctly.
OpenCore also displays the correct line order.
[2024-01-09_10-16_CLOVERX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/13874777/2024-01-09_10-16_CLOVERX64.EFI.log)

ybronst, 2024-01-10:
Hardware configuration:
CPU: i7-12700kf Alder Lake
MB: Asrock Z690 PG Reptide
RAM: 2x32Gb DDR4 3200 Crucial Technology
GPU: Radeon RX5700 Sapphire Nitro+ 8Gb
SSD: 1Tb WD_BLACK SN850X NVME M2 (PCI-e 4.0) - Mac OS 14.2.1/Windos 11
SSD: 521gb Samsung 860 EVO (SATA-6 Gbit) -
WIFI/BT: bcm943602CS in FV-AC1200 PCI-E adapter
BIOS: [v16.01](https://pg.asrock.com/mb/Intel/Z690%20PG%20Riptide/index.ru.asp#BIOS)
Display: ViewSonic 4K 60 Hz HDR via Display Port (VX3211-4K)

jief666, 2024-01-10:
Have you tried an older Clover to see if the order is correct ?

SergeySlice, 2024-01-11:
Set SMBIOS->Trust=true

SergeySlice, 2024-01-11:
Also set
SMBIOS->Memory->Channels=2
By default this value = 1

ybronst, 2024-01-11:
I can’t try Old Clover, since I don’t have it saved, and I couldn’t load either Ventura or Sonoma.
Bat I changed my config file like this:
`	<dict>
		<key>Memory</key>
		<dict>
			<key>Channels</key>
			<integer>2</integer>
		</dict>
		<key>Trust</key>
		<true/>
	</dict>`
and nothing has changed.
[image]
[image]
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/13907648/preboot.log)

SergeySlice, 2024-01-13:
Test please this version
[CLOVERX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13927187/CLOVERX64.efi.zip)
SMBIOS->Trust=true

ybronst, 2024-01-13:
> Test please this version [CLOVERX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13927187/CLOVERX64.efi.zip) SMBIOS->Trust=true
Again no change. Here are the results.
[image]
[image]
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/13928611/preboot.log)

ybronst, 2024-01-14:
I found my old preboot log, when the order of the dims was still determined correctly Clover revision: 5119 (HEAD, commit a0948b251).
Motherboard Asus Z77-a Memory Samsung 4096MB 1333MHz
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/13931628/preboot.log)

jief666, 2024-01-14:
It's not the same computer, so it's hard to compare and see what is now different in Clover (therefore find the bug).
Can you try to boot a Clover 5119 on your new computer ?

ybronst, 2024-01-14:
> It's not the same computer, so it's hard to compare and see what is now different in Clover (therefore find the bug). Can you try to boot a Clover 5119 on your new computer ?

Yes, this is my old computer, but later versions of Clover displayed the same memory error as the new one. Therefore, I did not update Clover on it, considering that the Mac OS would not be updated either.

jief666, 2024-01-15:
Slice did some modification regarding that, Try this clover : [CloverX64-2024-01-15-18-47-28-8697e93-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13942238/CloverX64-2024-01-15-18-47-28-8697e93-jief.zip).

ybronst, 2024-01-15:
> Slice did some modification regarding that, Try this clover : [CloverX64-2024-01-15-18-47-28-8697e93-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13942238/CloverX64-2024-01-15-18-47-28-8697e93-jief.zip).

Without changes.
![Снимок экрана 2024-01-15 в 22 18 58](https://g
[image]
ithub.com/CloverHackyColor/CloverBootloader/assets/156095944/77bafc60-84c6-4e0c-a5f8-39f6f898befa)
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/13942791/preboot.log)

jief666, 2024-01-27:
@YBronst Sorry, I but I think this fix is beyond my current skill... Let's see what Slice will do.

jief666, 2024-02-21:
It's possible by asking to do a lot of test and generates a lot of debug log. But since it's only cosmetic, I'm guessing it's a lot of time for something we almost never look at (system information).
Sorry.

YBronst, 2024-02-22:
> It's possible by asking to do a lot of test and generates a lot of debug log. But since it's only cosmetic, I'm guessing it's a lot of time for something we almost never look at (system information). Sorry.

Yes, this is a cosmetic problem. I would just like to bring Clover to the level of OpenСore.

jief666, 2024-02-23:
Yes, me too. But the quick look I took made me realize I know almost nothing about SMBIOS etc. So, In the time I have, I can't do it sorry.

YBronst, 2024-02-24:
> Yes, me too. But the quick look I took made me realize I know almost nothing about SMBIOS etc. So, In the time I have, I can't do it sorry.

Could you get Slice to fix it?