CloverHackyColor/CloverBootloader #661: Problem Boot

GiesteiraPG7, 2024-01-09:
boot only restarts with the latest commits

[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/13875857/preboot.log)

[https://we.tl/t-7hmioHo7Fr](url)

GiesteiraPG7, 2024-01-09:
it just boots at +++++ and restarts in a loop

GiesteiraPG7, 2024-01-09:
https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2815195

jief666, 2024-01-09:
But it works the last version I've sent you ?

GiesteiraPG7, 2024-01-09:
Nop

jief666, 2024-01-09:
Are you not PG7 on InsanelyMac ?

GiesteiraPG7, 2024-01-09:
Yes, but this is no longer atheros wifi!

here it's another machine and no longer boots with the latest versions of Clover in Commit

jief666, 2024-01-09:
Ok. So this compter, can you boot with an older Clover or an OpenCore ?

GiesteiraPG7, 2024-01-09:
just has r5156 (commit: 187400d0a) yes

jief666, 2024-01-09:
Ok.
Could you setup a repo with your EFI as described here : https://jief-machak.gitbook.io/cloverhelp/initial-setup
It'll make our life so much easier.

jief666, 2024-01-10:
I suggest you follow very carefully this guide : https://jief-machak.gitbook.io/cloverhelp/init
It'll be too slow otherwise.

jief666, 2024-01-15:
Does the problem still exist with this version : [CloverX64-2024-01-15-18-47-28-8697e93-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13942203/CloverX64-2024-01-15-18-47-28-8697e93-jief.zip) ?

GiesteiraPG7, 2024-01-15:
same concern
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/13942315/preboot.log)

jief666, 2024-01-15:
Do not send preboot.log. Instead, activate debug in boot section and you'll get log with file named like 2024-01-15_18-51_CloverX64.efi.log.
Send me that log.

GiesteiraPG7, 2024-01-15:
[2024-01-15_19-31_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/13942466/2024-01-15_19-31_BOOTX64.EFI.log)

jief666, 2024-01-15:
Much better !

Now, do you have a previous version of Clover that can boot that same compter ?

GiesteiraPG7, 2024-01-15:
> Much better !
> 
> Now, do you have a previous version of Clover that can boot that same compter ?

[2024-01-15_19-48_BOOTX64.EFI boot ok r5156.log](https://github.com/CloverHackyColor/CloverBootloader/files/13942558/2024-01-15_19-48_BOOTX64.EFI.boot.ok.r5156.log)

jief666, 2024-01-15:
Same goes for AppleXcpmCfgLock

GiesteiraPG7, 2024-01-16:
> AppleCpuPmCfgLock is true with the old version, but false with the latest. Do you use the same Clover folder, so we can be sure that the config.plist is identical ?
I use everything the same

jief666, 2024-01-16:
Oups, sorry, I must have compare the wrong files...

jief666, 2024-01-16:
I made a mistake thinking I made a mistake !!!

So no, I compared the right files. I'm seeing :

```
ACPCL 0 AXCL 0 AXEM 0 AXFB 0 CSG 0 DIM 1 DLJ 1 DRC 0 DPM 0 EBTFF 0 EDI 0 FAI 0 IPBS 0 LKP 0 PNKD 1 PTKP 1 TPD 1 XPL 0 PCC 0
C

```
versus

```
ACPCL 1 AXCL 1 AXEM 0 AXFB 0 CSG 0 DIM 1 DLJ 1 DRC 0 DPM 0 EBTFF 0 EDI 0 IPBS 0 LKP 0 PNKD 1 PTKP 1 TPD 1 XPL 0 PCC 0
G
```
ACPCL=AppleCpuPmCfgLock and AXCL=AppleXcpmCfgLock

jief666, 2024-01-16:
Ijust found that Clover will turn this patch on if MSR 0xE2 is locked.
In the log that doesn't work we can see : 
```
MSR 0xE2 before patch 7E008006
MSR 0xE2 is locked, PM patches will be turned on
```
And you know what ? I've heard of it but I don't know what that is !

GiesteiraPG7, 2024-01-16:
I'm going to test here with OpenCore v0.9.8 to see if the boot is normal and solve the "problem" if it is the CFG (MSR 0xE2)

jief666, 2024-01-16:
Clover check MSR 0xE2 way before calling OC. And in your previous log, it wasn't locked. So that's a change. I know it's not in the bios. But there is way to lock/unlock it.
So something changed. Did you change any driver. Wasn't there a driver to unlock MSR 0xE2 ?

GiesteiraPG7, 2024-01-16:
> Clover check MSR 0xE2 way before calling OC. And in your previous log, it wasn't locked. So that's a change. I know it's not in the bios. But there is way to lock/unlock it. So something changed. Did you change any driver. Wasn't there a driver to unlock MSR 0xE2 ?

The only change he made was Boot.efi and Cloverx64.efi, the rest is the same as what I'm using in version v5156

booted with OpenCore v0.9.8 exactly the same and it boots normally

I'll try to use a way to disable CFG and I'll be right back

[boot oc v0.9.8](https://snipboard.io/7xS3mC.jpg)

GiesteiraPG7, 2024-01-16:
all tests have the same result

jief666, 2024-01-16:
Here is a 5156 version, the exact same one that works, just with added logs.
Boot it and send me the log please.

GiesteiraPG7, 2024-01-16:
> Here is a 5156 version, the exact same one that works, just with added logs. Boot it and send me the log please.
[
](https://we.tl/t-qybwIQ9R9X)

jief666, 2024-01-16:
Forgot to send the efi file. Try this one : [CloverX64-2024-01-16-19-17-46-7e3ffed-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13954373/CloverX64-2024-01-16-19-17-46-7e3ffed-dirty-jief.zip)
and send me the log please. Send me ONLY the log. Thanks.

GiesteiraPG7, 2024-01-16:
> It's not completely finished ! Could try with this config : [config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13955089/config.plist.zip) and this Clover efi : [CloverX64-2024-01-16-13-35-00-7e3ffed-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13955115/CloverX64-2024-01-16-13-35-00-7e3ffed-jief.zip) ?

Now I'm home

tomorrow I test

jief666, 2024-01-16:
Sure, no pb.

GiesteiraPG7, 2024-01-17:
> It's not completely finished ! Could try with this config : [config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13955089/config.plist.zip) and this Clover efi : [...]

Work
[2024-01-17_12-17_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/13963083/2024-01-17_12-17_BOOTX64.EFI.log)

jief666, 2024-01-17:
Ok. so it does mean that you need these 2 settings KernelPM and AppleIntelCPUPM.
Leave them on (true), because the automatic activation of these 2 settings has been removed from Clover master branch.
This is still compatible with old clover because old clover just set them on for you. If they are already on, no pb of course.

Any further question on that topic ? If not, you're welcome to close this thread.

GiesteiraPG7, 2024-01-17:
> D'accord. cela signifie donc que vous avez besoin de ces 2 paramètres KernelPM et AppleIntelCPUPM. Laissez-les activés (true), car l'activation automatique de ces 2 paramètres a été supprimée de la branche principale de Clover. Ceci est toujours compatible avec le vieux trèfle, car le vieux trèfle vient de les installer pour vous. Si ils sont déjà allumés, pas de pb bien sur.
> 
> D'autres questions sur ce sujet ? Sinon, vous pouvez fermer ce fil.

everything is ok in this case, thank you for all the work done

GiesteiraPG7, 2024-01-17:
Thank you to the entire team and one in particular to @Jief666 for all the support he has given the community

sorted out

closed 2024-01-17 by GiesteiraPG7