CloverHackyColor/CloverBootloader #662: Error loading kernel (0x1) with MLPostFactor

Lost-Entrepreneur439, 2024-01-10:
I know very little people use MLPostFactor these days, especially on hackintoshes, but I still think this is an issue that should be fixed.

When attempting to boot OS X Mountain Lion 10.8.5 with MLPostFactor on my Dell Latitude D630 with the latest Clover, I get "Error loading kernel (0x1)". I believe this is due to MLPostFactor's modified kernel. The same 0x1 error also occurs in OpenCore. r5125 gives me this error, 
[image]
r5123 is the latest where I can boot without issues.

Lost-Entrepreneur439, 2024-01-10:
i forgot to add, here's my mach_kernel
[mach_kernel.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13891859/mach_kernel.zip)

SergeySlice, 2024-01-13:
Replace "boot" file with this one
 
[boot.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13927193/boot.zip)

Lost-Entrepreneur439, 2024-01-13:
With that, I got the error "boot0ss: error"

Lost-Entrepreneur439, 2024-01-14:
Turns out something went wrong with my USB, with that boot file, I get the 7 on boot, but then it just sits on a white line with no USB or SSD activity

SergeySlice, 2024-01-25:
"7"? For USB you should use "6" modification.

Lost-Entrepreneur439, 2024-01-25:
I don't know to be honest, I replaced the boot file and the 6 changed to a 7.

SergeySlice, 2024-02-24:
If you can't find good boot file take this one
[boot.zip](https://github.com/CloverHackyColor/CloverBootloader/files/14394012/boot.zip)