CloverHackyColor/CloverBootloader #663: [AMD AM5] RestrictEvents.kext not working

fabiosun, 2024-01-16:
I use in my hackintosh Restrictevents to have a correct name exposed in About my mac and to avoid misconfiguration dimm module error without using memory section of my config.plist
in Clover bootloader i see this:
[image]

and this for the DIMM which is correct using that kext:
[image]

kext is loaded:
`Last login: Tue Jan 16 17:28:23 on console
fabio@Pro-di-fabio-2 ~ % kextstat | grep -v com.apple
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
   58    4 0xffffff8003caa000 0x2f000    0x2f000    as.vit9696.Lilu (1.6.7) 087C34D2-49F8-3FDA-8757-27E9425C9EE4 <9 7 6 3 2 1>
   59    0 0xffffff8003cf1000 0x83000    0x83000    as.vit9696.WhateverGreen (1.6.7) 77B1BB94-033F-3D7D-810B-CB750E1B1362 <58 18 9 7 6 3 2 1>
   60    0 0xffffff80048ef000 0xc5000    0xc5000    com.dhinakg.AMFIPass (1.4.0) 919CD3B3-83DF-333E-99A2-00433847489A <58 9 8 7 6 3 2 1>
   61    0 0xffffff80049ee000 0xc000     0xc000     com.khronokernel.FeatureUnlock (1.1.6) B60AC2DF-42C5-3C2C-AD9B-A966369BA36B <58 9 7 6 3 2 1>
   62    0 0xffffff80049fa000 0xd000     0xd000     as.vit9696.RestrictEvents (1.1.4) 76F672D3-620C-3C18-B075-1AFFFE741684 <58 9 7 6 3 2 1>
   68    0 0xffffff8004a07000 0x8000     0x8000     com.osy86.USBWakeFixup (1) 227A2D50-9E8A-3B69-8BE1-1756705549FF <17 6 3>
   70    0 0xffffff8003cd9000 0x18000    0x18000    org.netkas.FakeSMC (3.5.6) 6CB12DBE-9E8E-39E5-905A-D9EE5DC3590E <17 9 7 6 3>
  177    3 0xffffff7f9620a000 0xb8ff2    0xb8ff2    org.virtualbox.kext.VBoxDrv (6.1.48) 23D43297-8004-34F4-8FE2-9FFB442D54BA <9 7 6 3 1>
  178    0 0xffffff7f962eb000 0x3ffc     0x3ffc     org.virtualbox.kext.VBoxNetFlt (6.1.48) 7B65B212-CF63-3F3A-A5BC-9C8188471EF4 <177 9 7 6 3 1>
  179    0 0xffffff7f962e5000 0x1ff3     0x1ff3     org.virtualbox.kext.VBoxNetAdp (6.1.48) 7B74659B-A224-3EF6-8841-32A822A9056A <177 7 6 1>
  181    0 0xffffff7f962f0000 0x4ffa     0x4ffa     org.virtualbox.kext.VBoxUSB (6.1.48) 6A2CF0B8-3E52-3574-A786-D25AFF03481B <180 177 75 9 7 6 3 1>
fabio@Pro-di-fabio-2 ~ % `
it doesnt change my CPU name ad in OpenCore it does

I am not sure if it is an user error or a bootloader error

Thank you to help also in this field :)

fabiosun, 2024-01-16:
this is hoew Opencore does in about my mac
[image]

jief666, 2024-01-16:
Are you you it's RestrictEvents.kext that does that ?
If you boot OpenCore without RestrictEvents.kext your processor name disappear ?

fabiosun, 2024-01-16:
Without it I see Xeon processor or unknown 
It depends also from a SMBIOS parameter callde ProcessorType, in opencore is a number value set to 0

fabiosun, 2024-01-17:
> If you boot OpenCore without RestrictEvents.kext your processor name disappear ?

[image]

this is the result without restrictevents
`Last login: Wed Jan 17 07:42:27 on console
fabio@Pro-di-fabio-2 ~ % kextstat | grep -v com.apple
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
   58    6 0xffffff8003caa000 0x2f000    0x2f000    as.vit9696.Lilu (1.6.7) 087C34D2-49F8-3FDA-8757-27E9425C9EE4 <9 7 6 3 2 1>
   59    0 0xffffff8003cf3000 0x83000    0x83000    as.vit9696.WhateverGreen (1.6.7) 77B1BB94-033F-3D7D-810B-CB750E1B1362 <58 18 9 7 6 3 2 1>
   60    0 0xffffff8003db8000 0xc000     0xc000     com.khronokernel.FeatureUnlock (1.1.6) B60AC2DF-42C5-3C2C-AD9B-A966369BA36B <58 9 7 6 3 2 1>
   61    0 0xffffff800496f000 0xc5000    0xc5000    com.dhinakg.AMFIPass (1.4.0) 919CD3B3-83DF-333E-99A2-00433847489A <58 9 8 7 6 3 2 1>
   62    2 0xffffff8003cd9000 0x1a000    0x1a000    as.vit9696.VirtualSMC (1.3.2) 2C500D02-E8EE-3A60-91A7-4AD28CD297E9 <58 17 9 7 6 3 1>
   63    0 0xffffff800493f000 0xd000     0xd000     as.lorys89.SMCProcessorAMD (1.0.0) 2BB30F15-23D9-34E8-936B-58B997020F27 <62 58 18 9 7 6 3 2 1>
   72    0 0xffffff8003db0000 0x8000     0x8000     com.osy86.USBWakeFixup (1) 227A2D50-9E8A-3B69-8BE1-1756705549FF <17 6 3>
   77    0 0xffffff800494c000 0x23000    0x23000    ru.joedm.SMCSuperIO (1.3.3) 484229E5-89FE-3D1E-B4F5-712949B0ACF3 <62 58 17 9 7 6 3 2 1>
  178    3 0xffffff7f9620a000 0xb8ff2    0xb8ff2    org.virtualbox.kext.VBoxDrv (6.1.48) 23D43297-8004-34F4-8FE2-9FFB442D54BA <9 7 6 3 1>
  179    0 0xffffff7f962eb000 0x3ffc     0x3ffc     org.virtualbox.kext.VBoxNetFlt (6.1.48) 7B65B212-CF63-3F3A-A5BC-9C8188471EF4 <178 9 7 6 3 1>
  180    0 0xffffff7f962e5000 0x1ff3     0x1ff3     org.virtualbox.kext.VBoxNetAdp (6.1.48) 7B74659B-A224-3EF6-8841-32A822A9056A <178 7 6 1>
  182    0 0xffffff7f962f0000 0x4ffa     0x4ffa     org.virtualbox.kext.VBoxUSB (6.1.48) 6A2CF0B8-3E52-3574-A786-D25AFF03481B <181 178 75 9 7 6 3 1>
fabio@Pro-di-fabio-2 ~ % 
`

fabiosun, 2024-01-17:
And this is Clover without restrictevents:
[image]

`Last login: Wed Jan 17 07:53:11 on console
fabio@Pro-di-fabio-2 ~ % kextstat | grep -v com.apple
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
   58    8 0xffffff8003caa000 0x2f000    0x2f000    as.vit9696.Lilu (1.6.7) 087C34D2-49F8-3FDA-8757-27E9425C9EE4 <9 7 6 3 2 1>
   59    0 0xffffff8003cf3000 0x83000    0x83000    as.vit9696.WhateverGreen (1.6.7) 77B1BB94-033F-3D7D-810B-CB750E1B1362 <58 18 9 7 6 3 2 1>
   60    0 0xffffff8004914000 0xc5000    0xc5000    com.dhinakg.AMFIPass (1.4.0) 919CD3B3-83DF-333E-99A2-00433847489A <58 9 8 7 6 3 2 1>
   61    0 0xffffff8004a13000 0xc000     0xc000     com.khronokernel.FeatureUnlock (1.1.6) B60AC2DF-42C5-3C2C-AD9B-A966369BA36B <58 9 7 6 3 2 1>
   62    1 0xffffff8004a3d000 0xd000     0xd000     com.ChefKissInc.RadeonSensor (1.3.0) F9729B63-6F09-39CB-B52D-6ACB7702D0D5 <58 18 13 7 6 3 2 1>
   63    3 0xffffff8003cd9000 0x1a000    0x1a000    as.vit9696.VirtualSMC (1.3.2) 2C500D02-E8EE-3A60-91A7-4AD28CD297E9 <58 17 9 7 6 3 1>
   64    0 0xffffff8004a1f000 0xd000     0xd000     as.lorys89.SMCProcessorAMD (1.0.0) 2BB30F15-23D9-34E8-936B-58B997020F27 <63 58 18 9 7 6 3 2 1>
   73    0 0xffffff8004a2c000 0x8000     0x8000     com.osy86.USBWakeFixup (1) 227A2D50-9E8A-3B69-8BE1-1756705549FF <17 6 3>
   78    0 0xffffff8003e44000 0x23000    0x23000    ru.joedm.SMCSuperIO (1.3.3) 484229E5-89FE-3D1E-B4F5-712949B0ACF3 <63 58 17 9 7 6 3 2 1>
   97    0 0xffffff8004a34000 0x9000     0x9000     com.ChefKissInc.SMCRadeonGPU (1.3.0) D96B51C6-038D-3F11-AAA6-B604420247E0 <63 62 58 18 13 9 7 6 3 2 1>
  181    3 0xffffff7f9620a000 0xb8ff2    0xb8ff2    org.virtualbox.kext.VBoxDrv (6.1.48) 23D43297-8004-34F4-8FE2-9FFB442D54BA <9 7 6 3 1>
  182    0 0xffffff7f962f0000 0x4ffa     0x4ffa     org.virtualbox.kext.VBoxUSB (6.1.48) 6A2CF0B8-3E52-3574-A786-D25AFF03481B <181 180 76 9 7 6 3 1>
  183    0 0xffffff7f962e5000 0x1ff3     0x1ff3     org.virtualbox.kext.VBoxNetAdp (6.1.48) 7B74659B-A224-3EF6-8841-32A822A9056A <181 7 6 1>
  184    0 0xffffff7f962eb000 0x3ffc     0x3ffc     org.virtualbox.kext.VBoxNetFlt (6.1.48) 7B65B212-CF63-3F3A-A5BC-9C8188471EF4 <181 9 7 6 3 1>`

jief666, 2024-01-17:
I guess it's time again to track any config.plist difference !
I'm guessing you're back using your previous config.plist with patches being a little different. ACPI was also different. Maybe SMBIOS was also different. ACPI, ACPI patches, kernel patches, loaded kexts, etc should be identical. And when I say identical, i mean IDENTICAL !
But of course, it's also possible that OC patches Cpu ids, memory, etc. differently than Clover. And maybe you'll need a config parameter in Clover but not in OC or vice versa.
And we already have another user having a similar problem.
Unfortunately, that's not my area of expertise and I hope @SergeySlice will take over that issue.

What I can do for you, in return of your help, is to clean your Clover configuration.
Send me the OC folder properly configured and a Clover folder. I'll update the Clover config based on OC.

fabiosun, 2024-01-17:
> Send me the OC folder properly configured and a Clover folder. I'll update the Clover config based on OC.

[problem restrict.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13961872/problem.restrict.zip)

Thank you for your help
only differene is in this declared kext  but this is another story i would like to understand with you :)

IO80211FamcilyLegacy.kext

Clover doesnt need to declare kext nested in a kext Plugin
Opencore needs it (to declare in config.plist i mean)

jief666, 2024-01-17:
> Clover doesnt need to declare kext nested in a kext Plugin
> Opencore needs it (to declare in config.plist i mean)
> 

Yes, this is the way it works. Just deleted plugins you don't want.

> hidden my data for MacPro 7.1 bios in both config files

What are you afraid of ? Because the problem is most likely SMBIOS, for little I know. So If you remove that... How are we gonna boot to try ?

fabiosun, 2024-01-17:
> How are we gonna boot to try ?

mo sfrsid ..doesnt is a custom way when people publish in public their data?

However system boots fine with both config i have sent and problem is the same
let me know if you need UUID i deleted

fabiosun, 2024-01-17:
> Yes, this is the way it works. Just deleted plugins you don't want.

for this i would have a question..but it is not related to this issue

jief666, 2024-01-17:
You just deleted an UUID ? Serial number ?

fabiosun, 2024-01-17:
Serial number UUID and MLB if i am not wrong at all

jief666, 2024-01-17:
The ACPI files are different. If you take the one from OC and put them into Clover, does it still boot ?

jief666, 2024-01-17:
> Serial number UUID and MLB if i am not wrong at all

Yes we should not need those.

fabiosun, 2024-01-17:
> The ACPI files are different.

mmmh explain they are the same here

jief666, 2024-01-17:
Sorry, my bad, comparison mistake. They are identical. Good.

fabiosun, 2024-01-17:
[image]

fabiosun, 2024-01-17:
if it could be useful , i found this
https://github.com/acidanthera/RestrictEvents/blob/master/RestrictEvents/RestrictEvents.cpp#L225

jief666, 2024-01-17:
Yes, I found it too yesterday. A kext runs AFTER kernel is booted. Therefore, OC or Clover has disappeared already.
So this has to do with cpuid patch made by the bootloader BEFORE.
So what cpuid do you have when OC is booted and when CLover is booted. They should be different.

fabiosun, 2024-01-17:
> So what cpuid do you have when OC is booted and when CLover is booted. They should be different.

sorry for delay in the answer
How to check and how to have them identical?

fabiosun, 2024-01-17:
i see this with this command in terminal:
`Last login: Wed Jan 17 12:16:47 on console
fabio@Pro-di-fabio-2 ~ % sysctl -a | grep brand
machdep.cpu.brand_string: AMD Ryzen 9 7950X 16-Core Processor            
machdep.cpu.brand: 0
fabio@Pro-di-fabio-2 ~ % 
`
but xeon on about my mac:
[image]

this with OpenCore without loading the kext

jief666, 2024-01-17:
Hey !! I shouldn't have to do everything !
If you just google "get cpuid from macos command line", second result says "sysctl -a | grep machdep.cpu"

So do this : sysctl -a | grep machdep.cpu
and let's compare the result.

jief666, 2024-01-17:
You can do this command using 
  OC without RestrictEvents.kext and then with it, so we'll know what RestrictEvents is patching.

fabiosun, 2024-01-17:
No applied kext
`Last login: Wed Jan 17 13:23:08 on console
fabio@Pro-di-fabio-2 ~ % sysctl -a | grep machdep.cpu
machdep.cpu.mwait.linesize_min: 64
machdep.cpu.mwait.linesize_max: 64
machdep.cpu.mwait.extensions: 3
machdep.cpu.mwait.sub_Cstates: 17
machdep.cpu.thermal.sensor: 0
machdep.cpu.thermal.dynamic_acceleration: 0
machdep.cpu.thermal.invariant_APIC_timer: 1
machdep.cpu.thermal.thresholds: 0
machdep.cpu.thermal.ACNT_MCNT: 1
machdep.cpu.thermal.core_power_limits: 0
machdep.cpu.thermal.fine_grain_clock_mod: 0
machdep.cpu.thermal.package_thermal_intr: 0
machdep.cpu.thermal.hardware_feedback: 0
machdep.cpu.thermal.energy_policy: 0
machdep.cpu.xsave.extended_state: 743 2432 2440 0
machdep.cpu.xsave.extended_state1: 15 2432 6144 0
machdep.cpu.arch_perf.version: 0
machdep.cpu.arch_perf.number: 0
machdep.cpu.arch_perf.width: 0
machdep.cpu.arch_perf.events_number: 0
machdep.cpu.arch_perf.events: 0
machdep.cpu.arch_perf.fixed_number: 0
machdep.cpu.arch_perf.fixed_width: 0
machdep.cpu.cache.linesize: 64
machdep.cpu.cache.L2_associativity: 8
machdep.cpu.cache.size: 1024
machdep.cpu.address_bits.physical: 48
machdep.cpu.address_bits.virtual: 48
machdep.cpu.tsc_ccc.numerator: 0
machdep.cpu.tsc_ccc.denominator: 0
machdep.cpu.max_basic: 16
machdep.cpu.max_ext: 2147483688
machdep.cpu.vendor: AuthenticAMD
machdep.cpu.brand_string: AMD Ryzen 9 7950X 16-Core Processor            
machdep.cpu.family: 25
machdep.cpu.model: 1
machdep.cpu.extmodel: 6
machdep.cpu.extfamily: 10
machdep.cpu.stepping: 2
machdep.cpu.feature_bits: 9140110466969500671
machdep.cpu.leaf7_feature_bits: 4055865257 4218830
machdep.cpu.leaf7_feature_bits_edx: 268435472
machdep.cpu.extfeature_bits: 8485406217078045695
machdep.cpu.signature: 10882834
machdep.cpu.brand: 0
machdep.cpu.features: FPU VME DE PSE TSC MSR PAE MCE CX8 APIC SEP MTRR PGE MCA CMOV PAT PSE36 CLFSH MMX FXSR SSE SSE2 HTT SSE3 PCLMULQDQ MON SSSE3 FMA CX16 SSE4.1 SSE4.2 MOVBE POPCNT AES XSAVE OSXSAVE AVX1.0 RDRAND F16C
machdep.cpu.leaf7_features: RDWRFSGS BMI1 AVX2 SMEP BMI2 ERMS INVPCID PQM PQE AVX512F AVX512DQ RDSEED ADX SMAP AVX512IFMA CLFSOPT CLWB AVX512CD SHA AVX512BW AVX512VL AVX512VBMI UMIP PKU GFNI VAES VPCLMULQDQ AVX512VNNI AVX512BITALG AVX512VPOPCNTDQ RDPID FSREPMOV L1DF
machdep.cpu.extfeatures: SYSCALL XD 1GBPAGE EM64T LAHF LZCNT PREFETCHW RDTSCP TSCI
machdep.cpu.logical_per_package: 32
machdep.cpu.cores_per_package: 16
machdep.cpu.microcode_version: 0
machdep.cpu.processor_flag: 1
machdep.cpu.core_count: 16
machdep.cpu.thread_count: 32
fabio@Pro-di-fabio-2 ~ % 
`

jief666, 2024-01-17:
I don't know how you copy/paste, but you'll have to find a way to keep the newline. This is too much of a nightmare to read...

jief666, 2024-01-17:
I still managed to compare them and there is NO dfference at all.

fabiosun, 2024-01-17:
> it looks like it's for intel cpu.

If you do not use RestrictEvents kext and set ProcessorType to 0 in SMBIOS OpenCore parameter CPU is seen as an Intel Xeon 16 cores
With restrictevents it use the name found here :
machdep.cpu.brand_string

> Could you use the DEBUG version of RestrictEvents.kext and activate the log ?
I would try to do this (never did in recent time) :)

fabiosun, 2024-01-17:
[OC-Lilu_1.6.7_23.3.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13965179/OC-Lilu_1.6.7_23.3.txt)
[Clover-Lilu_1.6.7_23.3.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13965180/Clover-Lilu_1.6.7_23.3.txt)

I hope you can find something here
By the way F11 seems to not work (not sure because i did many test to have a debug with some data in)

fabiosun, 2024-01-17:
[Clover-revproc-Lilu_1.6.7_23.3.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13965337/Clover-revproc-Lilu_1.6.7_23.3.txt)
[OC-revproc-Lilu_1.6.7_23.3.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13965338/OC-revproc-Lilu_1.6.7_23.3.txt)

more informations here

jief666, 2024-01-17:
In Clover-revproc-Lilu_1.6.7_23.3.txt, we can see :
`
requested to patch CPU name to AMD Ryzen 9 7950X 16-Core Processor 
`
and that is the same as in OC. So honestly I don't know why the patch fails. I don't even know what this is patching. Kernel ? Smbios ?
I'll have a look to your config.plist, make identical to OC and then we'll compare Smbios, ACPI, etc.

fabiosun, 2024-01-17:
[image]
Type paramater was useful in the past to have A correct name for CPU (it said Xeon xx core number) as it happens now in Opencore if you use ProcessorType parameter and without using Restrictevents
RestrictEvents helps to output the exact cpu name

fabiosun, 2024-01-17:
[image]
ok solved
Clover Needs a more complete SMBIOS compiled config.plist (i am verifying it)

fabiosun, 2024-01-17:
[image]
only this is needed

fabiosun, 2024-01-17:
@jief666 thank you for your time it was much simple and it was a problem of my Knowledge

closed 2024-01-17 by fabiosun

Z-fly, 2025-02-05:
> <img alt="image" width="466" src="https://private-user-images.githubusercontent.com/6358373/297469369-27c1e5bb-c53e-401c-966f-def26b9d3a84.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzg3NjMxNTcsIm5iZiI6MTczODc2Mjg1NywicGF0aCI6Ii82MzU4MzczLzI5NzQ2OTM2OS0yN2MxZTViYi1jNTNlLTQwMWMtOTY2Zi1kZWYyNmI5ZDNhODQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDIwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAyMDVUMTM0MDU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDVjYmQ3MWVkMDQ0YjljMWQwOGQ2OWQ0NGRmOGE2MjZiYWZjNWQ2ZmVmOWUyYmZlNzY0NmI0Mzk2OGM5YzdjZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.xx6kDfv12qlN8QHGsNb0QiWV8T0AwfX7vLVc2X7t-Qk">
> only this is needed

Thanks.