CloverHackyColor/CloverBootloader #666: The starting sound disappeared.

ybronst, 2024-01-18:
The starting sound disappeared. Clover revision: 5157 ALC897 used by AppleALC.kext layout-id: 0C000000

SergeySlice, 2024-01-19:
There is no Clover issue. May be you forgot something?

ybronst, 2024-01-19:
> There is no Clover issue. May be you forgot something?
What exactly could I have forgotten? In Release 5156 the startup sound still worked. The BlockSkywalk quirk did not work.

jief666, 2024-01-19:
Try this : [CloverX64-2024-01-19-15-43-13-64e7d98-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13990680/CloverX64-2024-01-19-15-43-13-64e7d98-dirty-jief.zip), please ?

ybronst, 2024-01-19:
> Try this : [CloverX64-2024-01-19-15-43-13-64e7d98-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13990680/CloverX64-2024-01-19-15-43-13-64e7d98-dirty-jief.zip), please ?

The sound appeared again, but when I try to select another output, instead of turning off the sound, it either freezes or suddenly reboots.

fabiosun, 2024-01-19:
Same result, it reboots when i choose another device

fabiosun, 2024-01-19:
[Archive.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13990950/Archive.zip)
in misc folder it prints these files

jief666, 2024-01-19:
And this : [CloverX64-2024-01-19-16-58-31-64e7d98-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13991391/CloverX64-2024-01-19-16-58-31-64e7d98-dirty-jief.zip) ?

fabiosun, 2024-01-19:
@jief666 thank you also for this tests
[Archive.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13991487/Archive.zip)
My system now in Option has two AMD audio sources (i cant see internal USB audio)
No error but no sound from hdmi/Dp devices

ybronst, 2024-01-19:
> And this : [CloverX64-2024-01-19-16-58-31-64e7d98-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13991391/CloverX64-2024-01-19-16-58-31-64e7d98-dirty-jief.zip) ?
No error but no sound from any audio outputs.

jief666, 2024-01-19:
Try this, please : [CloverX64-2024-01-19-17-56-58-64e7d98-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13991995/CloverX64-2024-01-19-17-56-58-64e7d98-dirty-jief.zip)

fabiosun, 2024-01-19:
No audio here
and selecting AMD audio as output after some tries it reboots

jief666, 2024-01-19:
You got audio with CloverX64-2024-01-19-16-58-31-64e7d98-dirty-jief.zip but not with CloverX64-2024-01-19-17-56-58-64e7d98-dirty-jief.zip ?

fabiosun, 2024-01-19:
No Jief my results are different i do not see internal audio device as ybronst did with a version
i have USB internal audio on my motherboard.
His audio is different
instead from GPU audio (HDMI/DP) maybe there is a problem for both

jief666, 2024-01-19:
@SergeySlice : instead of renaming gEfiAudioIoProtocolGuid, I completely disabled OcAudioLib.
Can you have a look to see if it works for you ?

ybronst, 2024-01-19:
> @SergeySlice : instead of renaming gEfiAudioIoProtocolGuid, I completely disabled OcAudioLib. Can you have a look to see if it works for you ?

Yes, everything is great.

closed 2024-01-19 by ybronst