CloverHackyColor/CloverBootloader #668: Clover doesn't appear, blinking dash.

jurandy007, 2024-01-24:
I have an Asus M5A78L-M USB 3 that works with Clover in version 5070, but in this version it doesn't have the timeout function, so I have to confirm it every time to log in (even if I change the config.plist > timeout file to "5", nothing happens).

I tried to update to version 5157, but in the usb clover boot only the blinking cursor appears and the Clover menu for choosing the system doesn't appear. Any fixes or solutions for this? any help is welcome.

SergeySlice, 2024-01-25:
The blinking cursor means that preloader "boot" can't be found. Check your files.

jurandy007, 2024-01-30:
Excuse me, but which files should I check?

SergeySlice, 2024-02-20:
Resolved

closed 2024-02-20 by SergeySlice

canob, 2025-02-08:
Hi @JurandyRafael , how are you?

Con you give me more details about how you manage to get version 5157 to boot up? Like you, I'm using version 5070 to boot a Manjaro Linux on a ASUS M5A88-M motherboard (Legacy BIOS), without any problem (only the "no timeout problem"), but with version 5157, I wait 20 seconds, and after that I get a Clover text interface (not the graphical one), and Clover cannot see my NVME disk that is on a PCI-E card when I list devices/partitions from that interface (on both versions I already copied the NvmExpressDxe.efi file to the corresponding folders).

Any advice?

Thanks in advance for your help.