CloverHackyColor/CloverBootloader #670: [Feature implementation Request] Add block kext feature in config.plist

fabiosun, 2024-01-25:
It's not really an issue but a request or my wish list for Clover :)
Clover has a very interesting possibility when you're at the boot menu level
Pressing the space button allows you to access this section:
[image]
[image]
[image]
It is also possible to lock in another section of the config.plist of OSX internal kexts as we do with SkywalkFamily:
[image]

@jief666 , @SergeySlice 

I'd like to understand if it's possible to implement kext locking that we have in the various folders in Clover/Kexts by editing the config.plist, as after all it's already possible to do at the bootmenu level.
Or does that possibility already exist?

YBronst, 2024-01-25:
As far as I know, such a similar function is available in the OС because all kexts there are stored in a common folder, so each aded kext is entered into the config in a certain order, and the ability has been added not to delete a line, but to disable it. 
which is not very convenient if there is no way boot to desktop.
Unlike the OS, a problematic kext can be blocked once by a space or permanently moved to the off folder, which can be done even from the Clover shell.
In Clover, unlike the OS, a problematic kext can be blocked once with a space or permanently moved to a disabled folder, which can be done even from the Clover shell. Therefore, I guess that this function will be duplicate.

fabiosun, 2024-01-25:
> permanently moved to a disabled folder

i do not need to disable it permanently but only when i have to update Sonoma
As you know it is not possible to use Delta update if you have patched Sonoma with OCLP (OpenCore Legacy patcher)
so my request or wishlist if you prefer

I would like to change my config .plist and not the entire EFI or as i wrote, changing with a proper config i could disable only the kext i need to disable, configuring in it proper SIP and all features i need to use Apple delta OS update
then once uon a time i have updated the system and patched again with OCLP to use my previous confg.plist with all the kexts i need for sonoma
I hope is more clearer for you my request
to say in a shorter way:

1) with OCLP applied delta Update in sonoma is not availabe because SIP and others mandatory oprion useful to OCLP
2) preparing a good config-1.plist with all parameters useful for OS delta update as SIP, No kext blocked or all i need for a NO OCLP installation and in this way update the system
3) use a proper config.plist to use OCLP to start root patching (in my case only for Modern network patching)
4) end

i know and i wrote above how clover can do with boot menu block kext option, but it could be faster if it was possible only changing a config.plist as we can do in Clover now

YBronst, 2024-01-25:
> > permanently moved to a disabled folder
> 
> i do not need to disable it permanently but only when i have to update Sonoma As you know it is not possible to use Delta update if you have patched Sonoma with OCLP (OpenCore Legacy patcher) so my request or wishlist if you prefer
> 
> I would like to change my config .plist and not the entire EFI or as i wrote, changing with a proper config i could disable only the kext i need to disable, configuring in it proper SIP and all features i need to use Apple delta OS update then once uon a time i have updated the system and patched again with OCLP to use my previous confg.plist with all the kexts i need for sonoma I hope is more clearer for you my request to say in a shorter way:
> 
> 1. with OCLP applied delta Update in sonoma is not availabe because SIP and others mandatory option useful to OCLP
> 2. preparing a good config-1.plist with all parameters useful for OS delta update as SIP, No kext blocked or all i need for a NO OCLP installation and in this way update the system
> 3. use a proper config.plist to use OCLP to start root patching (in my case only for Modern network patching)
> 4. end
> 
> i know and i wrote above how clover can do with boot menu block kext option, but it could be faster if it was possible only changing a config.plist as we can do in Clover now

For what? I moved the kexts and disabled them, updated the system, patched it, returned the kexts and everything was OK. I don’t even do this by hand, but through a command file.

fabiosun, 2024-01-26:
Hello @YBronst ,
Forgive me but I don't quite understand the meaning of these posts of yours
Do you think this request of mine is already there or is it unnecessary?
If it is present as I am requesting it, can you show it to me?
If you think it is useless, it is your opinion and as such I respect it ;)

YBronst, 2024-01-26:
> Hello @YBronst , Forgive me but I don't quite understand the meaning of these posts of yours Do you think this request of mine is already there or is it unnecessary? If it is present as I am [...]
I'm sorry, but you don't understand.
You are trying to offer a conceptually incompatible solution from OpenCore to Clover.
The only thing common to both boot loaders is the OpenRuntime driver.
Developers will never agree to greater integration and this is not a question of technical feasibility, but a question of historically formed program ideology.
I can't even convince the developers to move the CsrActiveConfig parameter from the RtVariables section to the SMBIOS section. I'm sorry! ;)

fabiosun, 2024-01-26:
> You are trying to offer a conceptually incompatible solution from OpenCore to Clover.

Is it possible with Opencore to choose a different config.plist?
I do not think so and i am not comparing or asking for a greater integration! :)

Clover has inside this features
My request is to use them in config.plist as we can do for patches, ssdt and so on (enabling / disabling them in boot menu)

offtopic here:
Your request to move CsrActiveConfig in SMbios section for you is usefull but you cand do with a different config or manually chanching SIP values in Bootmenu Option

> I'm sorry! ;)

Why? :)
let the devs to express their thoughts about this whishlist if they have some spare time to read all this :) :)

YBronst, 2024-01-26:
Slice has already expressed his opinion about this on applelife.ru :)

fabiosun, 2024-01-26:
so i have to hope in Jief for an answer :PP

YBronst, 2024-01-26:
Slice and I previously discussed my request to block kexts, but he only did BlockSkywalk.
He told me that this is not necessary, since there is a folder that Clover does not process.

fabiosun, 2024-01-26:
> He told me that this is not necessary, since there is a folder that Clover does not process.

Let see
It seems a bit different, one OSX only. Sonoma needs are different for delta update using OCLP

YBronst, 2024-01-26:
Yes, I understand this, but it is all the more difficult to implement. Most likely, this will not be done through the config. Why? Here's why: https://www.applelife.ru/threads/clover.42089/page-1538#post-1045185
"The story goes like this: Asindanthera saw Clover as a useful technique. After scratching my head a little, we did something similar, but in a completely different way.
Then people come to us with a baggage of axioms from Dortania, and begin to ask how to do this in Clover.
You don’t need this in Clover, you’d better read KCH.
I don’t have the strength to write a site similar to Dortanya; in this they are undoubtedly ahead. Not everything is right there, but it is useless to fight the parishioners.
For them, this is a bible that needs to be bowed to the waist." :(

fabiosun, 2024-01-26:
i dont read Dortania but Configuration.pdf
Not interested in a war about Clover and OpenCore
I use both
For this reason i am asking this "feature"
If I didn't use OCLP (OpenCoreLegacy Patcher for My wifi in SONOMA) I would not have opened this issue/whishlist
@jief666 solved recently a big proble for AM5 platform so...i can hope!
 :)

jief666, 2024-01-26:
@YBronst 
> I can't say that this will never happen, but OpenCore is written in C, and Clover is written in C++.
> I'd love to do this myself, but I don't know enough C++.

Clover is _compiled_ by C++, but 90% of it is still C.

> CsrActiveConfig from RtVariables to SMBIOS.

Any technical reason ?

@fabiosun 
I'm not sure to understand the request. You want to block an extension prelinked into the kernel ?

fabiosun, 2024-01-26:
> @fabiosun
> I'm not sure to understand the request. You want to block an extension prelinked into the kernel ?

No jief, thank you for your answer
i would like to block using a config.plist mode kexts i use in Kext/other
[image]

And this is possible to do in boot menu pressing space and block kext

jief666, 2024-01-26:
Hum, like YBronst, I'm not sure to understand why you want this. Because changing of config.plist from the menu seems the same as untick a kext from that menu.
Sorry to say that. I'm judging the features you'd like to have. But just draw your attention to something you may not have thought of.
And because that wouldn't be something quick to add, and Slice seems against it, I'm not too sure you'll see that soon.
But maybe you can convince us...

fabiosun, 2024-01-26:
> why you want this
I use OCLP (opencore legacy patcher) to use my wireless/bt in macOS Sonoma
OCLP needs for this:

1) sip disabled
2) block a kext
3) adding some old kexts in kext/other

In this condition is not possible to update macOS via delta update (i update my system often)

So in Opencore I create a second EFi with:
1)SIP Enabled
2) unblock the kext
3)disabling old kexts (untick in config.plist)
I choose this with your Bootloader chooser
clear Nvram
Start Sonoma and i do the delta update from system/Software update

restart the system
relaunch OCLP (opencore legacy patcher)
and then with Bootloader chooser 
i use the EFi with sip disabled and kext blocked and old Kexts injected in Kext folder.

In Clover is more simple (but it could be simpler with my whishlist feature)

I have only an EFi with 2 config.plist
one useful to update so with sip enabled, unblocked kext and with old kext in CLOVER/Kext folder which i untick in bootmenu option

One EFI  useful to start root patching with  OCLP with the specs i said before for opencore bootloader

So to have it working well with clover i do this steps
1) change config.plist to config-1.plist
2) reset Nvram
3) untick manually those kext prelinked one and the two in Other folder

from here my wish list
It seems from a no expert programmer point of view  Clover has this "feature" , it is only not exposed and available in config.plist

obviusly i am not able to understand if this is a big job to do 
I have only asked to you (Slice and you) if this opportunity is there or i too much complex to do 

> But maybe you can convince us...

I cannot do better of above lines! :)

jief666, 2024-01-27:
> block a kext
> adding some old kexts in kext/other

Which one ?

> delta update

What is "delta update" ? The normal update process from inside macOS ?

> In this condition is not possible to update macOS via delta update (i update my system often)

Shouldn't we instead find why you can't update ? Is this a know unsolvable problem ?

> 2) unblock the kext

Which one ?

Do you know that there is a feature I did a long time ago (yeah, I know, we don't really have a documentation)  which allow you to enable kext only for install or recovery.
If you want a kext to only be enabled for Sonoma install, just put it in a folder called 14_install. If it's for recovery, use 14_recovery. If you want a kext be used for only "normal" boot, use 14_normal. Any kext in 14 will be used for all boots.
NOTE : it's even possible to use folder like 14.1, 14.1_normal, 14.1_install, 14.1_recovery if you need a kext for a very specific version. You can even use 14.1.2 for a very very specific version.

fabiosun, 2024-01-27:
> Which one ?

IO80211FamilyLegacy.kext
IOSkywalkFamily.kext

> What is "delta update" ? The normal update process from inside macOS ?

Yes from here:
[image]

> Shouldn't we instead find why you can't update ? Is this a know unsolvable problem ?

I think is a known problem when people use some config.plist parameter (SIP,SecureBoot ) and OCLP Root Patching
I think also is "unsolved"

> Which one ?

[image]

and this is yet possible to uncheck via config.plist

> Do you know that there is a feature I did a long time ago (yeah, I know, we don't really have a documentation) which allow you to enable kext only for install or recovery.

No and it seems a very good undocumented feature :)
But it seems to me that it can't satisfy my wishlist or am I wrong?

basically, without the root patch running the OCLP app, I can't use WIFI in macOS Sonoma
The patch allows (by breaking the operating system seal) to blockSkywalk, as we can now also do properly in CLover bootloader and use the macOS Ventura kexts useful for wifi
these kexts mentioned just above:
IO80211FamilyLegacy.kext
IOSkywalkFamily.kext

so, the solution of having two config.plist seems to me the only one at the moment
Or continue as I am doing
I prepare the two configs with the options I can already do , i.e. SIP, BlockSkywalk, and then manually change the config during the boot menu and complete what is needed by pressing space and then the option to block kexts in the Clover/Kext/xxx folder

thank you again for your listening

PS I am trying everything to convince you :) :)
Am I succeeding? :)

jief666, 2024-01-27:
If I understood well, you need to add this 2 kexts for wifi :
- IO80211FamilyLegacy.kext
- IOSkywalkFamily.kext
but doing so prevent you from updating.

Again, if I understood well, you'd like to block these 2 kexts and unblock the original one to be able to update. Am I right ?

fabiosun, 2024-01-27:
Yes :)
but internal OSX kext is yet possible to block and to unblock
see here
https://github.com/CloverHackyColor/CloverBootloader/issues/670#issuecomment-1913137648

jief666, 2024-01-27:
> but internal OSX kext is yet possible to block and to unblock

prelinked kext (the one on macOS drive) can be block or unblock. The one in Clover kext folder cannot. The idea is : if you don't need a kext, you put it in there.

> so, the solution of having two config.plist seems to me the only one at the moment

Yes, I can't see how you would get around that.

Do you have to restore original IOSkywalkFamily to be able to see the update ? I'm still not sure when you need to restore the original IOSkywalkFamily.

What about 2 Clover folders and BootloaderChooser ? Wouldn't that be the simplest anyway ?

fabiosun, 2024-01-27:
> Do you have to restore original IOSkywalkFamily to be able to see the update ? I'm still not sure when you need to restore the original IOSkywalkFamily.

it is an OCLP procedure (root patching and revert it when you have to update the system)
And it is not that..also differences in SIp and i SecureBootModel

> What about 2 Clover folders and BootloaderChooser ? Wouldn't that be the simplest anyway ?

Yes in this way it is the same i do in Opencore bootloader
I thought with option we have in Clover Bootloader at the boot menu stage this was possible to achieve in a different way with only 2 config.plist and 1 EFI

I will go on to do this how i do know
press space bar and untick stuff i cant untick in config.plist with a plist editor

YBronst, 2024-01-27:
Okay, you block these kexts, but how will you download delta updates? Over a wired connection? Or do you have a USB WiFi stick? If via an apvo connection, then I don’t see a problem with temporarily moving old kexts to the off folder. Then there is no need to fence the garden.

fabiosun, 2024-01-27:
@YBronst with the original Sonoma ones

fabiosun, 2024-01-27:
> then I don’t see a problem with temporarily moving old kexts to the off folder.

Probably i cant transfer well my thoughts 
or you do not use OCLP to patch modern wifi wireless in SOnoma

YBronst, 2024-01-27:
To begin with, you placed the old kexts in the wrong place.
Еhe old kexts should be in the folder 14 from which they are loaded only in Sonoma.
RestrictEvents.kext solves problem  the new unknown processors and number of memory banks, with SMBIOS Mac Pro. It can be in the  folder others and you don’t need to disable it for delta updates.

I don’t understand what doesn't give you just do this?

1. Сhange config.plist to config-1.plist;
2. Disabling old kexts (by pressing space);
3. Start macOS;
3. Do the delta update from system/Software update and just restart the system (when she asks);

If only Wi-Fi is not working, then no one forbids to you connect a USB Wi-Fi dongle stick, or wired connection.

There is no need to edit config, or reset the NVRAM.

Before Shabbat, I regularly updated Sonoma and Ventura that way. :)

I have 3 physical drives, each of which has its own set of operating systems and bootloaders. The first one has Windows and Sonoma + Clover installed. On the Second Ventura + Clover. On the third there is only file storage + OpenCore for emergency cases.
I asked Slice to move the CsrActiveConfig parameter from the RtVariables section to the SMBIOS section, which could reduce the number of Clovers, but was refused.
I have already found how to do this myself, even without being a great expert in C++ and C but this will take some time. If this modification will be completed, I'll make my own fork of Clover.

jief666, 2024-01-27:
> it is an OCLP procedure (root patching and revert it when you have to update the system)
> And it is not that..also differences in SIp and in SecureBootModel

I'm still not sure : you have to revert the root patching before downloading the update ?

fabiosun, 2024-01-27:
> > it is an OCLP procedure (root patching and revert it when you have to update the system)
> > And it is not that..also differences in SIp and in SecureBootModel
> 
> I'm still not sure : you have to revert the root patching before downloading the update ?

YES

fabiosun, 2024-01-27:
> To begin with, you placed the old kexts in the wrong place.
> Еhe old kexts should be in the folder 14 from which they are loaded only in Sonoma.
> RestrictEvents.kext solves problem the new unknown [...]

Please read well what i have written and please do not a lesson to me
i use only Sonoma
and i use kext in Other
and, i have no problem
if in a future i will have i create again the right folder for me

fabiosun, 2024-01-27:
> But in this case, your WiFi will turn off.

yes as it shoul be because sonoma does not support wifi with broadcom and fenvi card (and others)
Tou are akways saying about Sip and move it in other place
here is a bit offtopic
i change SIP easily in my second config.plist

jief666, 2024-01-27:
@fabiosun 
I understand you'd like only one Clover folder. Copies are a pain to sync.
Having only one Clover folder makes you go the the menu to change the config.plist, right ?

fabiosun, 2024-01-27:
yes

fabiosun, 2024-01-27:
@YBronst 
Or, or can you tell me and teach me how to do delta i OSX updates (no updates with a full installer but full) having OCLP installed
Or it is useless for you to repeat these things to me
Without any controversy
My initial question was mainly to see if I was doing something wrong or if it was a feature already available

we now understand that it is not available
The methods suggested in this peaceful discussion I know and apply them

YBronst, 2024-01-27:
> Please read well what i have written and please do not a lesson to me i use only Sonoma and i use kext in Other and, i have no problem if in a future i will have i create again the right folder for me

If you are stupid and don’t know the purpose of folders in Clover, then you don’t need to blame others for being stupid. Folder 14 is intended for those kexts that may turn out to be problematic in Sonoma, so that it is easier to block them by space, and others are placed in the folder that have been verified.
In short, I am categorically against the innovations you propose. This may break the kext injection logic.

fabiosun, 2024-01-27:
> Having only one Clover folder makes you go the the menu to change the config.plist, right ?

double Yes
And this is a unique Clover Bottloader feature

jief666, 2024-01-27:
@YBronst I think you should not use that tone...

jief666, 2024-01-27:
@fabiosun Ok, so currently, you have to go in the menu to change config.plist and untick the 2 kexts you added in Other. Is it that correct ?

YBronst, 2024-01-27:
> @YBronst I think you should not use that tone...

I am very sorry. He just annoyed me with his stubbornness. I will try to be correct in my statements.

fabiosun, 2024-01-27:
I don't seem to have offended either you or jief who is trying to understand my request

Whether or not you are in favor of my proposal doesn't change that much
Since it's not law and it's just a wish list

whether or not there will be this news I will continue to use Clover

and you will probably continue to ask for something that they rejected and interfere in a discussion that wasn't controversial without your intervention

i love you anyway
Shalom

jief666, 2024-01-27:
@YBronst That for the apology. Please do not answer when you are annoyed.

fabiosun, 2024-01-27:
yes

jief666, 2024-01-27:
So that feature would be to avoid you having to untick 2 kexts in a menu, but that you would still have to go into the Clover menu.

YBronst, 2024-01-27:
> @YBronst That for the apology. Please do not answer when you are annoyed.

Fine! I didn’t like the way you started arguing with me. I won't answer anymore. But I am against bringing options from OpenCore into Clover.

jief666, 2024-01-27:
> Fine! I didn’t like the way you started arguing with me. I won't answer anymore. But I am against bringing options from OpenCore into Clover.

I never said I agreed. I'm trying to understand exactly. I also learn about this thing about IOSkyWalk that I didn't know few days ago.

fabiosun, 2024-01-27:
> So that feature would be to avoid you having to untick 2 kexts in a menu, but that you would still have to go into the Clover menu.

sorry maybe i have not understood well
if feature is there 
i would press O and then only change the config to the right one (sip, and others) and the press Fkey to clear nvram (this to change the sip to the right value)

jief666, 2024-01-27:
And if the feature is not there, you would press O and then change the config to the right one (sip, and others), then untick the 2 kexts and the press Fkey to clear nvram (this to change the sip to the right value).

I don't think this feature would change much for you. It'll save you 2 unticks in the kexts menu.

fabiosun, 2024-01-27:
> I don't think this feature would change much for you. It'll save you 2 unticks in the kexts menu.

unticking could be usuful if you have also other kext to untick as ie here:
[image]
this is a real Mac Pro 5.1 i am trying to use in it your chooser and clover (and i am dfailing for now)
I will close this issue and sorry for your tie lost ;)

jief666, 2024-01-27:
@fabiosun Don't close. It's fine. Let's try to understand. There is something you don't know or something I don't know. Let's try to find out.
I don't know what it the software on this screenshot. When I say untick kexts, I'm saying untick that in Clover menu. Do you know that you can untick kexts from the Clover menu ?

fabiosun, 2024-01-27:
Do you mean as i show in originate post in first message of this issue?

> Don't close. It's fine. Let's try to understand.

I don't like misunderstandings even though this time I don't seem to have said anything
one can agree or disagree with opinion without "bombarding" the other with a "stupid" epithet

YBronst, 2024-01-27:
@fabiosun Don't close. 

> > I don't think this feature would change much for you. It'll save you 2 unticks in the kexts menu.
> 
> unticking could be usuful if you have also other kext to untick as ie here: <img alt="image" width="1106" src="https://private-user-images.githubusercontent.com/6358373/300191484-959c1b7c-b1e9-4a6e-a695-4739d339d3e3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYzNzM5MzEsIm5iZiI6MTcwNjM3MzYzMSwicGF0aCI6Ii82MzU4MzczLzMwMDE5MTQ4NC05NTljMWI3Yy1iMWU5LTRhNmUtYTY5NS00NzM5ZDMzOWQzZTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMjdUMTY0MDMxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDhiZTA2ODM0NGQ0YzZjYWY1N2I3ZjBkOWM2ZTZkZGNhYzc4NDc0NDNlZGFjNGExYzA5Zjg2ZDhhNTczZjY2OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.USopg9gVxdbdDmGOtUOKKSyx37uBBbIlktotHkz8yUQ"> this is a real Mac Pro 5.1 i am trying to use in it your chooser and clover (and i am failing for now) I will close this issue and sorry for your time lost ;)

This may be helpful for understanding. As for the real Mac, I can assume that the failure was due to the order of the kexts in the table. Here is the right order that allowed my old Mac Pro 6.1 to boot into Sonoma.

jief666, 2024-01-27:
> Do you mean as i show in originate post in first message of this issue?

Yes, what's wrong with that ? Because we know that you have to go into some Clover menu anyway.

jief666, 2024-01-27:
What's this editor you are using for config.plist ?

fabiosun, 2024-01-27:
> was due to the order of the kexts in the table.

See  @YBronst , that's what I expect from you being an experienced user.
Not the personal attacks <3

On that mac I can't run sonoma because I have mounted an RX6600 graphics card which is not supported by OCLP for the time being
In a Hack it works well also in sonoma
In Mac Pro 5.1 only till Monterey (it is a CPU feature list).
but here a bit off topic
i try to convince @jief666 using all my arrows :)

@jief it i a graphical plist Editor (OCAT) it is only OpenCore compatible

YBronst, 2024-01-27:
> What's this editor you are using for config.plist ?

I usually use PlistEdit Pro.app, but in the case of the OC - OCAuxiliaryTools.app

YBronst, 2024-01-27:
> See @YBronst , that's what I expect from you being an experienced user. Not the personal attacks <3
>
> On that mac I can't run sonoma because I have mounted an RX6600 graphics card which is not [...]

Never mind, I wasn't offended. And don't be offended by me.
But here I don’t agree with you. Radeon RX 6650 XT works fine for me in Mac pro 6.1 and in other hacks.

jief666, 2024-01-27:
There is 2 conversations going on at the same time at that doesn't make it easy.
YBronst : you seems to speak about disable kexts by modifying your Clover folder prior to reboot.
It seems to me that fabiosun would like to be able to reboot without having to modify his clover folder.

jief666, 2024-01-27:
Let's not add a conversation about graphic cards, please.

YBronst, 2024-01-27:
> There is 2 conversations going on at the same time at that doesn't make it easy.
> YBronst : you seems to speak about disable kexts by modifying your Clover folder prior to reboot.
> It seems to me that fabiosun would like to be able to reboot without having to modify his clover folder.

Probably yes!

jief666, 2024-01-27:
@fabiosun so far, it seems to me that this feature would only save you to go in the kext Clover menu. Am I correct ?

fabiosun, 2024-01-27:
> t seems to me that fabiosun would like to be able to reboot without having to modify his clover folder.

> It seems to me that fabiosun would like to be able to reboot without having to modify his clover folder.

correct
but also correct order could be a good point to discuss :)
with those two kexts i would like to untick we have a great fortune in Clover
Inside one of these kext there is a plugin folder
with inside another kext
Clover add automatically in the right order and all the patches work
with opencore we have to declare the correct order (please dont get it wrong i am not comparing two booloaders)

[27 more replies]

closed 2024-02-17 by fabiosun