CloverHackyColor/CloverBootloader #673: Clover r5157 not boot

Makeroot911, 2024-02-08:
Clover r5156 can boot, but clover r5157 can't booting only +++++++ and restart
[preboot 5156 work.log](https://github.com/CloverHackyColor/CloverBootloader/files/14215585/preboot.5156.work.log)
[preboot 5157 not work.log](https://github.com/CloverHackyColor/CloverBootloader/files/14215587/preboot.5157.not.work.log)

SergeySlice, 2024-02-10:
As I see you are using release version from 12.01.2024. Some bugs already corrected since that. Test please this version

[CLOVERX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/14228480/CLOVERX64.efi.zip)

SergeySlice, 2024-02-10:
And some comments about drivers:
~~~~
0:119  0:000  === [ LoadDrivers ] =============================
0:120  0:000  Loading FSInject.efi status=Success
0:122  0:002  Loading NvmExpressDxe.efi status=Success
0:125  0:002   - driver needs connecting
0:125  0:000  Loading EnglishDxe.efi status=Success
0:127  0:002  Loading AptioInputFix.efi status=Success
0:130  0:002  Loading DataHubDxe.efi status=Success
0:132  0:002  Loading PartitionDxe.efi status=Success
0:135  0:002   - driver needs connecting
0:135  0:000  Loading ApfsDriverLoader.efi status=Success
0:137  0:002   - driver needs connecting
0:137  0:000  Loading AudioDxe.efi status=Success
0:140  0:002   - driver needs connecting
0:140  0:000  Loading VirtualSmc.efi status=Success
0:244  0:103  Loading VBoxExt2.efi status=Success
0:246  0:002   - driver needs connecting
0:246  0:000  Loading VBoxExt4.efi status=Success
0:249  0:002   - driver needs connecting
0:249  0:000  Loading Fat.efi status=Success
0:251  0:002   - driver needs connecting
0:251  0:000  Loading VBoxHfs.efi status=Success
0:254  0:002   - driver needs connecting
0:254  0:000  8 drivers needs connecting ...
0:254  0:000  PlatformDriverOverrideProtocol->GetDriver overriden
0:254  0:000  Partition driver loaded: 
0:254  0:000  HFS+ driver loaded
0:254  0:000  APFS driver loaded
0:254  0:000  Searching for invalid DiskIo BY_DRIVER connects: not found, all ok

~~~~

1.  FSInject.efi - used only for 10.7 Lion.
2. EnglishDxe.efi - useful for first UEFI like H61. Your UEFI already contains it.
3. DataHubDxe.efi - embedded into Clover since 5140
4. PartitionDxe.efi - used for Apple formatted disks 10years ago.
5. VirtualSmc.efi  - I affraid it is no more compatible with Clover. Use FakeSMC instead.
6. VBoxExt2.efi, VBoxExt4.efi - check your Linux installation what FS it uses EXT2 or EXT4.

macfoxys, 2024-02-10:
Same problem with r5157.
Try last version from post  still reboot after ++++++.
Try change apito drivers - no changes 

[preboot 5156 BOOT.log](https://github.com/CloverHackyColor/CloverBootloader/files/14229196/preboot.5156.BOOT.log)
[preboot 5157 REBOOT.log](https://github.com/CloverHackyColor/CloverBootloader/files/14229197/preboot.5157.REBOOT.log)

Makeroot911, 2024-02-10:
@SergeySlice still same with the last version 
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/14229538/preboot.log)

SergeySlice, 2024-02-22:
> @SergeySlice still same with the last version [preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/14229538/preboot.log)

Stop after ++++++ mostly always because of locked MSR 0xE2.
I see in your preboot.log
~~~~
0:100  0:000  MSR 0xE2 before patch 1E008006
0:100  0:000  MSR 0xE2 is locked, PM patches will be turned on
~~~~
So you must set in config.plist KernelPM=true

macfoxys, 2024-02-22:
> Stop after ++++++ mostly always because of locked MSR 0xE2. I see in your preboot.log
>
> ```
> 0:100  0:000  MSR 0xE2 before patch 1E008006
> 0:100  0:000  MSR 0xE2 is locked, PM patches will be turned on [...]

thanks, but after the change  config.plist  no reboot, boot it freezes after ++++++
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/14379139/preboot.log)
[image]

SergeySlice, 2024-02-24:
I am not sure if FakeSMC 1800 is compatible with Clover because new Clover included SMChelper.efi which is tuned for FakeSMC v3.5.x [FakeSMC3_with_plugins ](https://github.com/CloverHackyColor/FakeSMC3_with_plugins)

Second...
~~~~
24:058  0:000  No OpenRuntime driver. This is ok, OpenRuntime is not mandatory.
~~~~

Oh, no! Use please OpenRuntime.efi v1.2

macfoxys, 2024-02-24:
> I am not sure if FakeSMC 1800 is compatible with Clover because new Clover included SMChelper.efi which is tuned for FakeSMC v3.5.x [FakeSMC3_with_plugins [...]

Oh, yes! owersome, it's worked!!!
Спасибо, дружище!!!

SergeySlice, 2024-02-24:
А я-то думал, там реальная проблема!

closed 2024-02-24 by SergeySlice

macfoxys, 2024-02-24:
> А я-то думал, там реальная проблема!

)) ну, кому, как) редко обновляю Кловер, драйверы, работает да и ладно, а тут Сонома и BCM94360, пришлось обновлять Кловер, на ОС очень переходить не хотелось.
обычно при обновлении проблем со старыми кончиками не было. В любом случае, спасибо за помощь и за Кловер, 
хотя совокупность проблем таки была.