CloverHackyColor/CloverBootloader #676: Failure boot Clover-5152 ?

Acid0hacK, 2024-02-27:
[image]
Who know whats happening ? After create with BDU bootable flash drive, booting only this and i don't see standard graphic interface.
See only this interface, like on screen.

MB: Gigabyte P55-USB3
CPU: i7-870

SergeySlice, 2024-03-02:
It means you booted llegacy way BIOS->MBR->PBR->boot (CloverEFI) and see the screen. But next step should be /EFI/CLOVER/CLOVERX64.EFI which is absent.

sambow23, 2024-03-03:
I'm having the same issue but CLOVERX64.efi is present 
[image]

SergeySlice, 2024-03-06:
Then you booted one drive while Clover located on other drive.

sambow23, 2024-03-12:
> Then you booted one drive while Clover located on other drive.

It's booted directly off the USB with no other drives in the system. There is only 1 FAT32 partition with clover on it

SergeySlice, 2024-03-30:
I know the screen from topic starter. It is an interface of boot6 (reduced DUET) and it appears if CLOVERX64.EFI is not found.
You may navigate through the menu and see where is the file.

closed 2024-03-30 by SergeySlice