CloverHackyColor/CloverBootloader #677: General Protection Fault on Dell R720

ikogan, 2024-02-28:
Recently I started encountering a GPF in UEFI on at least 2 Dell R720s with BIOS 2.9.0:

[image]

I'm trying to boot a Proxmox on a Samsung SSD 950 PRO NVMe drive using this Sabrent adapter: https://www.amazon.com/gp/product/B084GDY2PW.

This was working for quite some time and randomly failed on my existing server today. It's also failing the same way on another R720 that wasn't being used until today. This stack trace happens both when attempting to boot from the NVMe and when doing simple things like "Print all UEFI boot options to log". I'm trying Clover 5157.

k0lja, 2024-03-08:
Encounter the same Problem in UEFI on 2 R640s with BIOS 2.20.1. Also trying to run proxmox with a similar m2 nvme adapter.

dascathea, 2024-03-11:
Hi sir,

Did you ever solve this issue? I have Proxmox running on samsung 960 pro for a while without issues. I upgraded some rams and then the system was still booting fine. Then I plugged back my midplane drives, now it doesnt boot.

ikogan, 2024-03-11:
Nope, I ended up moving my OS to a normal SSD and using my NVMe as just another local-lvm thin pool.

loop1dev, 2024-03-13:
I ran into the same issue with a T620. I found that rEFInd worked after adding the nvme driver.

Armynator, 2024-03-15:
Same problem on a R720xd. Downgrading to some random older version (Clover 5132) worked.

brandonw62, 2024-03-24:
Encountered the same exact error on my dell R620 running BIOS version 2.9.0. Ran Boot Disk Utility (BDU) with clover version 5156 and am back up and running again. Here is a download link for BDU: https://www.softpedia.com/get/System/Boot-Manager-Disk/Bootdisk-Utility.shtml

titou10titou10, 2024-03-29:
Same problem here. R720 Bios 2.9.0, Clover v5157

SergeySlice, 2024-04-20:
I will wait for some essential observation or investigation.

ikogan, 2024-04-20:
What are some observations or investigation that could be helpful? I have an R720 that hasn't been configured yet that I might be able to do some work with if that one has this issue as well.

SergeySlice, 2024-05-06:
> What are some observations or investigation that could be helpful? I have an R720 that hasn't been configured yet that I might be able to do some work with if that one has this issue as well.

I wish you find what to report other than "not working".

ikogan, 2024-05-11:
> > What are some observations or investigation that could be helpful? I have an R720 that hasn't been configured yet that I might be able to do some work with if that one has this issue as well.
> 
> I wish you find what to report other than "not working".

I'm a little confused, maybe we have a misunderstanding as I feel like there's a good bit more detail in this thread than that. To summarize:

- I own a few Dell R720s.
- I am attempting to boot them off a Samsung SSD 950 PRO NVMe drive using a PCI-e adapter. The BIOS on these Dells does not support NVMe boot.
- The operating system installed on the NVMe drive was Proxmox, running a Linux 6.5 kernel, I can't remember the exact version. Probably something like 6.5.11.
- I have installed Clover on to a USB 3.0 flash drive
- Suddenly, and with no changes to the system, I started getting the error in the screenshot taken at the start of this post, a `General Protection Fault (13)` with the attached trace when booting that machine.
- This was happening both on a server that had been working and on a new one I hadn't yet setup.
- I also get the same error if I try to "Print all UEFI boot options to log" in the Clover UI. This was done with Clover version 5157.

Unfortunately at this time I have found a use for my remaining R720 but may be getting another in the future. Should I get one, is there other information that I could provide that would help this out? Again, for a full stack dump please see the screenshot in the first post.

SergeySlice, 2024-05-22:
I am very sorry but where did you see "Print all UEFI boot options to log" in the Clover UI.?
[image]

ikogan, 2024-05-22:
It's the option shown in the "Boot Options" screen here: https://github.com/5T33Z0/Clover-Crate/blob/main/GUI/Boot_Menu_Options.md#clover-boot-options:

[image]

NachtRaben, 2024-07-10:
Encountered this problem on an R630 and had to hard power cycle the server. Turned out to be an issue with the UEFI not communicating with the lifecycle controller. Unsure if related or coincidence that I encountered the same exact fault message.

jcastro, 2024-10-18:
Same here on R720. With Clover 5112 it boots but the partition is hidden (I need to press F3). Not sure what's new on latest clover but the partitions were showing up automatically

wcb33, 2024-10-31:
same problem on R720XD

clouserw, 2024-12-19:
I'm getting a similar error with a slightly different stacktrace on an r730xd.  This is using two samsung pro 990 1tb nvmes on a supermicro AOC-SLG3-2M2 with 4x4 pci slot bifurcation.

The crash references DellNetworkProtocol.efi, but all the PXE boot options are disabled in the BIOS if that matters.  Please let me know if I can provide more details.

[image]

OCR'd:

> A system restart is required. BIOS rev 2.19.0
> The system detected an exception during the UEFI pre-boot environment.
> 
> Press ENTER on the serial console (@115200 8-N-1) for detailed exception info.
Type: General Protection Fault (13) Source: Software (UEFI0011)
AX=AFAFAFAFAFAFAFAF BX=0000000070B605A8 SI=0000000076AB6DB0 DI=0000000076AB6D80
CX=0000000070B605AB DX=0000000000000000 R8=0000000000000000 R9=0000000000000000
BP=0000000076AB6DB0 SP=0000000076AB6D80 IP=00000000717BA04E Flags=00010286
(CurrentTPL = 10, TSC ticks since last event 00000018BE35)
> 
> LBR not available
-->rip 717BA04E Unknown (hkxqg) +07004E <-- Crash occurred here
s00 70B58198 DellNetworkProtocol.efi +3B8B6C
Stack Unwind Complete after 1 Frames

clouserw, 2024-12-19:
For all the folks here having the same problem, I did get this to work by reverting to Clover v5130.  So, something is broken between those two releases in my case.  In v5130 it no longer errors, but I run into https://github.com/CloverHackyColor/CloverBootloader/discussions/644 and have to drop to the clover shell and type:

```
FS1:
cd EFI/BOOT
BOOTX64.EFI
```

and then it boots right up.

jwidess, 2025-02-13:
I too was having the same issue trying to boot to WS 2025 on my R720. I followed this [YT Guide](https://www.youtube.com/watch?v=wqBa-eAnkCw) on how to setup the Clover drive via [Bootdisk Utility](https://cvad-mac.narod.ru/index/bootdiskutility_exe/0-5), however it was using the latest version of Clover (5161) which would cause the same error as @ikogan and can be seen below.

[image]

Someone with a [similar issue  on Reddit](https://www.reddit.com/r/homelab/comments/1d7s8fy/comment/l759th4/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button) discovered using an earlier version of Clover, in my case 5122, fixed the problem. To do this I used BDUtility but in the _**Options->Configuration**_ menu I selected "**Local**" instead of "DL" and placed the "**Clover-5122-X64.iso**" I downloaded and extracted from the [Github release here](https://github.com/CloverHackyColor/CloverBootloader/releases/download/5122/Clover-5122-X64.iso.7z) into the same location as BDUtility.exe. 

[image]

Then I proceeded to format the USB drive and copy the NVMe Driver just like the original video and voila, problem gone, boots into Windows fine!

digiphaze, 2025-11-26:
This is certainly tied to a bios version on the dells. I've had clover booting from the same NVME for years on my R730.  I just upgraded to BIOS 2.19.0 and immediately ran into this issue though with a slightly different crash location and "unknown(veg)" something.  Seems a problem with cold booting, warm booting it gets to clover. Maybe a race condition with the life cycle controller?