CloverHackyColor/CloverBootloader #681: where is my mistake

Nicomehr, 2024-03-23:
Hello
Background info: I have an older workstation. NVMe doesn't work. 2 M.2 Samsung SM951 are installed and are bootable, as well as 2 Samsung 970 EVO (not bootable due to NVMe)
I wanted to test faster M.2 with CLOVER and created a stick with the Clover 5157.
Since it does NOT work on the main PC, I tested it on the laptop.
The stick boots and several drives are displayed, two with somewhat cryptic characters. If I select the first cryptic drive, the installed Windows starts on the Samsung 970 Evo. The stick seems to work for me.
The same stick on the main PC boots and 8 drives are displayed (two Samsung SM951 each with “system reserved” + “C:” + “D:” partition and two Samsung 970 for data (displayed is HD1, HD2, HD3 , HD1, HD2, HD3 HD2, HD2))
I can choose whatever I want to start. A black window appears with a flashing cursor. That's it
Where is the mistake?

SergeySlice, 2024-03-30:
It looks like nothing to boot from these drives.
Old workstation means there is no UEFI BIOS? So you have to boot in legacy mode. Yes, you started the Clover in this mode but you can't start OS in the mode. You should reinstall Windows in UEFI mode provided by Clover. Make sure the drive formatted as GPT.

SergeySlice, 2024-04-19:
Abondoned.

closed 2024-04-19 by SergeySlice