CloverHackyColor/CloverBootloader #691: XCODE14 Clover compil error due to use of undeclared identifier 'TIMEOUTFP' in ~/CloverBootloader/Library/OsxBdsPlatformLib/BdsPlatform.c

matxpa, 2024-05-13:
@SergeySlice

Hi Slice

"./buildme XCODE14" (with different commits since 20240511, ... F.Y.I : local CloverBootloader repo is uptodate)

Compil error due to use of undeclared identifier 'TIMEOUTFP' in ~/CloverBootloader/Library/OsxBdsPlatformLib/BdsPlatform.c

8) build Clover (with XCode) --> XCODE14 (Monterey constraint (XCODE 14.2 max))

...
Running edk2 build for CloverX64 using the command:
build --cmd-len=50000 -D NO_GRUB_DRIVERS_EMBEDDED --conf=/~/CloverBootloader/Conf -D USE_LOW_EBDA -p Clover.dsc  -a X64 -b RELEASE -t XCODE14 -n 5 

Build environment: macOS-12.7.5-x86_64-i386-64bit
Build start time: 19:22:25, May.13 2024

WORKSPACE        = /~/CloverBootloader
EDK_TOOLS_PATH   = /~/CloverBootloader/BaseTools
CONF_PATH        = /~/CloverBootloader/Conf
PYTHON_COMMAND   = python3

Processing meta-data .
Architecture(s)  = X64
Build target     = RELEASE
Toolchain        = XCODE14
...
[SLINK] CustomizedDisplayLib
[CC] AutoGen
make: Nothing to be done for `tbuild'.
/~/CloverBootloader/Library/OsxBdsPlatformLib/BdsPlatform.c:1295:35: error: use of undeclared identifier 'TIMEOUTFP'
  PlatformBdsDiagnostics (IGNORE, TIMEOUTFP, BaseMemoryTest); //Slice - I want silent boot but with Memory test
                                  ^
/~/CloverBootloader/Library/OsxBdsPlatformLib/BdsPlatform.c:1307:13: error: use of undeclared identifier 'TIMEOUTFP'
  Timeout = TIMEOUTFP;
            ^
2 errors generated.
make: *** [/~/CloverBootloader/Build/Clover/RELEASE_XCODE14/X64/Library/OsxBdsPlatformLib/PlatformBds/OUTPUT/BdsPlatform.obj] Error 1

build.py...
 : error 7000: Failed to execute command
	make tbuild [/~/CloverBootloader/Build/Clover/RELEASE_XCODE14/X64/Library/OsxBdsPlatformLib/PlatformBds]

build.py...
 : error F002: Failed to build module
	/~/CloverBootloader/Library/OsxBdsPlatformLib/PlatformBds.inf [X64, XCODE14, RELEASE]

- Failed -
Build end time: 19:22:39, May.13 2024
Build total time: 00:00:14

Regards

SergeySlice, 2024-05-13:
This is macros defined in PlatformBds.inf
~~~~
[BuildOptions]
# TIMEOUTFP=0 -- silent boot
  XCODE:*_*_*_CC_FLAGS = -DTIMEOUTFP=0
  GCC:*_*_*_CC_FLAGS = -DTIMEOUTFP=0
  MSFT:*_*_*_CC_FLAGS = -DTIMEOUTFP=3

~~~~

chris1111, 2024-05-14:
No issue building from  Xcode Version 15.4 beta (15F5021i)
macOS Sonoma 14.5 (23F79)

[image]

[image]

matxpa, 2024-05-16:
> Répondre à : CloverHackyColor/CloverBootloader ***@***.***>

Hi

Are you realy sur what I see, on your first screenshot,

is 

Running edk2 build for Clover64 using the command:

build --cmd-len=50000 -D NO_GRUB_DRIVERS --conf=/Users/chris/src/CloverBootloader/Conf -D USE_LOW_EBDA -p Clover.dsc -a X64 -b RELEASE -t XCODE8 -n 5”

 

XCODE8 is NOT XCODE15

 

Regards

Screenshot.2024-05-13.at.8.27.54.PM.png (view on web)

Screenshot.2024-05-13.at.8.27.19.PM.png (view on web)

chris1111, 2024-05-16:
My point is not about a specific version of Xcode but rather to show you that compiling Clover with Xcode has no problem
See default toolchain from ebuild.sh

[image]

I don't use buildme,I never use it I use basic commands in macOS Sonoma 14 the latest and I will show you that it also works in Monterey 12 with Xcode 14.3.1

[image]

[image]

[image]

[image]

[image]

So it work from Monterey to Sonoma

closed 2024-05-18 by matxpa