CloverHackyColor/CloverBootloader #693: Autoboot Linux volume

Demontager, 2024-05-29:
My BIOS does not allow booting from NVME drive installed in pci-e slot.
I need to load Ubuntu 22.04 automatically. I just copied sample Clover config https://github.com/Clover-EFI-Bootloader/clover/blob/master/CloverPackage/CloverV2/EFI/CLOVER/config-sample.plist and edited only Boot section.
Getting volume name inside booted Ubuntu
```
Disk /dev/nvme0n1: 931.51 GiB, 1000204886016 bytes, 1953525168 sectors
Disk model: Netac NVMe SSD 1TB                      
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: A0F472E7-7C29-467C-83CE-CAF41F85C6EC

Device           Start        End    Sectors  Size Type
/dev/nvme0n1p1    2048       4095       2048    1M BIOS boot
/dev/nvme0n1p2    4096    1054719    1050624  513M EFI System
/dev/nvme0n1p3 1054720 1953523711 1952468992  931G Linux filesystem

$ blkid
/dev/nvme0n1p3: LABEL="ubunturoot" UUID="72ac49f8-b38c-4f17-88a9-7a0d2b558d72" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="650bb479-2ffd-4154-aebc-f86f8c230bb3"
```
Then added "ubunturoot" to DefaultVolume

```
	</dict>
	<key>Boot</key>
	<dict>
		<key>Arguments</key>
		<string>-v arch=i386 slide=0 darkwake=0</string>
		<key>Arguments</key>
		<string>kext-dev-mode=1 rootless=0</string>
		<key>DefaultLoader</key>
		<string>boot.efi</string>
		<key>DefaultVolume</key>
		<string>ubunturoot</string>
		<key>NeverHibernate</key>
		<false/>
		<key>Fast</key>
		<false/>
		<key>Legacy</key>
		<string>PBR</string>
		<key>LegacyBiosDefaultEntry</key>
		<integer>0</integer>
		<key>Debug</key>
		<false/>
		<key>Timeout</key>
		<integer>5</integer>
		<key>XMPDetection</key>
		<string>-1</string>
	</dict>
```

I been inspired by that post which seems worked https://github.com/foxlet/macOS-Simple-KVM/issues/566

SergeySlice, 2024-05-31:
You set DefaultLoader as "boot.efi". It is not your case. May be "grub.efi" will be more correct?

SergeySlice, 2024-06-21:
resolved

closed 2024-06-21 by SergeySlice