CloverHackyColor/CloverBootloader #694: Sierra installer can't fit itself into screen?

Randrianasulu, 2024-06-04:
I tried to use mix of pre-build config for my HP probook 4730s and self-build clover binary (because another pre-built one was also failing with red text expection)

https://github.com/ubihazard/probook-4x30s - there is legacy clover/efi folder I used.

[image]

[image]

setting "inject intel" does not improve situation, setting dual-link property as described on this github page also does not help.

Selecting "Inject ati" resulted in X64 expection. Using premade config OS-X-Clover-Laptop-Config-master/config_HD4000_1600x900.plist from https://codeload.github.com/RehabMan/OS-X-Clover-Laptop-Config/zip/refs/heads/master collection alse result in expection.

Randrianasulu, 2024-06-04:
[preboot-hp4730.txt](https://github.com/user-attachments/files/15546431/preboot-hp4730.txt)

preboot log (for some reason dumping tables via f4 hang cliver)

Randrianasulu, 2024-06-04:
[install.log](https://github.com/user-attachments/files/15550398/install.log)
[system.log](https://github.com/user-attachments/files/15550399/system.log)

install and system logs from Installer, I can't see anything bad specifically about gfx, apart from profile setting failure?

[displaypolicyd.log](https://github.com/user-attachments/files/15550413/displaypolicyd.log)

chris1111, 2024-06-05:
Try this EFI folder for installing Sierra from USB install media

[EFI.zip](https://github.com/user-attachments/files/15571767/EFI.zip)

Let me known?

Randrianasulu, 2024-06-05:
@chris1111 it seems to stop with bunch of acpi errors

[image]

chris1111, 2024-06-05:
it not stop  wait !!!!!

chris1111, 2024-06-05:
wait until the end it will go to the macOS installer

chris1111, 2024-06-05:
If you have successfully installed macOS, tell me and I will prepare a perfect post install for your HP Probook

Randrianasulu, 2024-06-05:
@chris1111 not yet ... I noticed first config.plist I used has this section:

```
<dict>
                                        <key>Comment</key>
                                        <string>change GFX0 to IGPU</string>
                                        <key>Find</key>
                                        <data>
                                        R0ZYMA==
                                        </data>
                                        <key>Replace</key>
                                        <data>
                                        SUdQVQ==
                                        </data>
                                </dict>
```

while yours missing it (and I waited for 10 min and gui not arrived - so I rebooted into win10 and run mbr2gpt ... hope it will boot after that! (not like I need windows perse ... but I hope to see most tools working as intended ..))

chris1111, 2024-06-05:
do you have another graphics card on this laptop

Randrianasulu, 2024-06-05:
@chris1111 techically yes, it does have switchable graphics (intel/ati) - but I tried to disable it in bios ... and this does not improve situation.....

bfitzgit23, 2024-06-05:
> If you have successfully installed macOS, tell me and I will prepare a perfect post install for your HP Probook

hi chris, i have a hp probook 4530s with no dgpu only the igpu. and it has a whitelist on it so im sadly stuck with the ar9285 stock wifi card and cant swap it. and my dad nor i have the desire nor know how to change it NOR mod the bios to enable another wifi card without putting the laptop into a bricked state beyond repair. I also don't know hackintosh enough to make it work without someone else's support tbh. So can you please help me man. Thank you. I also dont know how to do it on any machine outside the probook either tbh.

Randrianasulu, 2024-06-05:
[acpi-laptop.zip](https://github.com/user-attachments/files/15573415/acpi-laptop.zip)

Ah, dumper worked from Clover, just need to wait few seconds. 

It probably must be noted even Windows 10 booted in 1024*768 by default, and win 8.1 while it was alive also refused to set native 1600*1200 until I re-enabled switchable graphics. Linux (5.6, 6.1.xx) set console fine

closed 2024-06-05 by chris1111

chris1111, 2024-06-05:
> hi chris, i have a hp probook 4530s with no dgpu only the igpu. and it has a whitelist on it so im sadly stuck with the ar9285 stock wifi card and cant swap it. and my dad nor i have the desire nor [...]

what BIOS revision you have?

Randrianasulu, 2024-06-05:
@chris1111 I fixed it (at least screen now properly centered) by using fake id for SNB:

0x00020000

PS: may be moving this to discussions is right action?

Randrianasulu, 2024-06-05:
this plist
[config-1600x900.txt](https://github.com/user-attachments/files/15595500/config-1600x900.txt) now 
working for me (no QE/CI but at least native resolution!)

trackpad kinda wild, so usb mouse was used.

Randrianasulu, 2024-06-06:
ah, interesting. Previous plist was working grom usb (because I set it for legacy boot, by installing bootsectors manually) but when I copied it to purely EFI partition on hard drive even bootloader screen was scrambled! I re-instanced resolution 1024*768, disabled load vbios and re-enabled dual-link. Now it boots ok, just in 1024*768.

Randrianasulu, 2024-06-06:
interesting ... from running system it seems that this weird shifted mode IS native 1600*900 (vram for example jumps to 384mb instead of 3, if I remove my wrong fake id - I mixed up platform id and device id..) but it clearly does not fit.

so I can run in 4 modes:

1) bios based with 1024*768 unaccelerated fb
2) bios based with vbios loading and 1600*900 mode unaccelerated fb.
3) UEFI based boot with 1024*768 unaccelerated fb.
4) UEFI based boot with 1600*900 shifted/half-visible internal display (but apparently accelerated?)

Randrianasulu, 2024-06-06:
tried 

https://github.com/p00ya/displaymode/releases

but all modes ( 1024*768, 800*600)   shifted heavily to the right, with black border on the left side of internal  monitor .. I guess some binary patching needed ? :(