CloverHackyColor/CloverBootloader #696: Two buffer overflow write in AppleImageCodec

wjqsec, 2024-06-05:
I found two buffer overflow/arbitrary write in AppleImageCodec.

1.
In file CloverBootloader/Protocols/AppleImageCodec/picopng.c,  info->height and info->width are read from the image, however, if they are too large, it will allocate zero memory. Without a null-pointer check, it will use zero data pointer. In bootloader, zero-pointer deference does not lead to crash, it will reuse these zero pointers.  
[image]
[image]
[image]
these are payloads which can trigger the vulnerability.

2.
in file CloverBootloader/Protocols/AppleImageCodec/load_bmp.c, RealPixelWidth is indirectly calcualted from the image, however, if it is less than 2, it will cause a large loop in code:

for (x = 0; x <= RealPixelWidth - 2; x += 2)
      {
        ImageValue = *ImagePtr++;

        Index = ImageValue >> 4;
        PixelPtr->Blue = BmpColorMap[Index].Blue;
        PixelPtr->Green = BmpColorMap[Index].Green;
        PixelPtr->Red = BmpColorMap[Index].Red;
        PixelPtr->Reserved = AlphaValue;
        PixelPtr++;

        Index = ImageValue & 0x0f;
        PixelPtr->Blue = BmpColorMap[Index].Blue;
        PixelPtr->Green = BmpColorMap[Index].Green;
        PixelPtr->Red = BmpColorMap[Index].Red;
        PixelPtr->Reserved = AlphaValue;
        PixelPtr++;
      }
[payload_00597.log](https://github.com/user-attachments/files/15597084/payload_00597.log)
[payload_00741.log](https://github.com/user-attachments/files/15597085/payload_00741.log)
these two payloads can trigger the bmp decode vulnerabilities.

SergeySlice, 2024-06-08:
1. Show the code.
2. if x will be signed then there will be no large loop.

wjqsec, 2024-06-08:
1. For example, in function PNG_decode, line 1184, scanlines = vector8_new(((info->width * (info->height * bpp + 7)) / 8) + info->height, 0); in line 1192, vector8_resize(info->image, outlength); // time to fill the out buffer, the return values are not checked.
2. yes.

SergeySlice, 2024-06-09:
1. I committed some security checks. (look file header for the authors of the sources)
2. I really see negative length so no sense to check. We must use the value.

wjqsec, 2024-06-10:
ok, they look good.