CloverHackyColor/CloverBootloader #702: How to display hidden drives, which do not require pressing F3 to display them

wangtieying, 2024-06-30:
I have this situation now, which is that my motherboard cannot start the ESXi system under UEFI, but it can be started through Clover. However, starting under Clover does not automatically display and requires pressing F3 every time
And they can't start the system on their own yet

SergeySlice, 2024-07-05:
Show your config.plist and preboot.log.

wangtieying, 2024-07-06:
> 显示您的 config.plist 并preboot.log。

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Boot</key>
	<dict>
		<key>DefaultLoader</key>
		<string>\EFI\BOOT\BOOTX64.efi</string>
		<key>DefaultVolume</key>
		<string>ESP</string>
		<key>NeverHibernate</key>
		<false/>
		<key>Timeout</key>
		<integer>3</integer>
		<key>XMPDetection</key>
		<integer>-1</integer>
	</dict>
	<key>GUI</key>
	<dict>
		<key>Custom</key>
		<dict>
			<key>Entries</key>
			<array>
				<dict>
					<key>Disabled</key>
					<false/>
					<key>Hidden</key>
					<false/>
					<key>Path</key>
					<string>\EFI\BOOT\BOOTX64.efi</string>
					<key>Title</key>
					<string>ESXi</string>
					<key>Type</key>
					<string>Linux</string>
					<key>Volume</key>
					<string>881FC8C9-3564-458E-85E5-25723697EFE7</string>
					<key>VolumeType</key>
					<array>
						<string>Firewire</string>
						<string>Optical</string>
						<string>Internal</string>
						<string>External</string>
					</array>
				</dict>
				<dict>
					<key>Disabled</key>
					<false/>
					<key>Hidden</key>
					<false/>
					<key>Path</key>
					<string>\EFI\BOOT\BOOTX64.efi</string>
					<key>Title</key>
					<string>vmware</string>
					<key>Type</key>
					<string>Linux</string>
					<key>Volume</key>
					<string>ESP</string>
					<key>VolumeType</key>
					<array>
						<string>Firewire</string>
						<string>Optical</string>
						<string>Internal</string>
						<string>External</string>
					</array>
				</dict>
			</array>
		</dict>
		<key>Mouse</key>
		<dict>
			<key>Enabled</key>
			<true/>
			<key>Speed</key>
			<integer>8</integer>
		</dict>
		<key>Scan</key>
		<dict>
			<key>Entries</key>
			<true/>
			<key>Kernel</key>
			<string>None</string>
			<key>Legacy</key>
			<string>First</string>
			<key>Linux</key>
			<true/>
			<key>Tool</key>
			<true/>
		</dict>
		<key>ScreenResolution</key>
		<string>1024x768</string>
	</dict>
	<key>Quirks</key>
	<dict>
		<key>ResizeAppleGpuBars</key>
		<integer>-1</integer>
	</dict>
</dict>
</plist>

wangtieying, 2024-07-06:
``<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Boot</key>
	<dict>
		<key>DefaultLoader</key>
		<string>\EFI\BOOT\BOOTX64.efi</string>
		<key>DefaultVolume</key>
		<string>ESP</string>
		<key>NeverHibernate</key>
		<false/>
		<key>Timeout</key>
		<integer>3</integer>
		<key>XMPDetection</key>
		<integer>-1</integer>
	</dict>
	<key>GUI</key>
	<dict>
		<key>Custom</key>
		<dict>
			<key>Entries</key>
			<array>
				<dict>
					<key>Disabled</key>
					<false/>
					<key>Hidden</key>
					<false/>
					<key>Path</key>
					<string>\EFI\BOOT\BOOTX64.efi</string>
					<key>Title</key>
					<string>ESXi</string>
					<key>Type</key>
					<string>Linux</string>
					<key>Volume</key>
					<string>881FC8C9-3564-458E-85E5-25723697EFE7</string>
					<key>VolumeType</key>
					<array>
						<string>Firewire</string>
						<string>Optical</string>
						<string>Internal</string>
						<string>External</string>
					</array>
				</dict>
				<dict>
					<key>Disabled</key>
					<false/>
					<key>Hidden</key>
					<false/>
					<key>Path</key>
					<string>\EFI\BOOT\BOOTX64.efi</string>
					<key>Title</key>
					<string>vmware</string>
					<key>Type</key>
					<string>Linux</string>
					<key>Volume</key>
					<string>ESP</string>
					<key>VolumeType</key>
					<array>
						<string>Firewire</string>
						<string>Optical</string>
						<string>Internal</string>
						<string>External</string>
					</array>
				</dict>
			</array>
		</dict>
		<key>Mouse</key>
		<dict>
			<key>Enabled</key>
			<true/>
			<key>Speed</key>
			<integer>8</integer>
		</dict>
		<key>Scan</key>
		<dict>
			<key>Entries</key>
			<true/>
			<key>Kernel</key>
			<string>None</string>
			<key>Legacy</key>
			<string>First</string>
			<key>Linux</key>
			<true/>
			<key>Tool</key>
			<true/>
		</dict>
		<key>ScreenResolution</key>
		<string>1024x768</string>
	</dict>
	<key>Quirks</key>
	<dict>
		<key>ResizeAppleGpuBars</key>
		<integer>-1</integer>
	</dict>
</dict>
</plist>

wangtieying, 2024-07-06:
> 显示您的 config.plist 并preboot.log。

[config.txt](https://github.com/user-attachments/files/16116527/config.txt)

wangtieying, 2024-07-06:
> 显示您的 config.plist 并preboot.log。

[preboot.log](https://github.com/user-attachments/files/16116529/preboot.log)

avenger49, 2025-04-05:
I have the same problem. Installed Proxmox on M.2 and have to hit F3 after the Clover screen shows up and have 2 options, one is and white icon with SS and the other is the same icon but without the SS. If I click on the icon without the SS, I get a black screen, but after about 5 or 10 seconds, if I hit the Enter key, I got about a 40% chance of it booting into Proxmox, otherwise it just sits there on a black screen.

SergeySlice, 2025-07-21:
> > 显示您的 config.plist 并preboot.log。
> 
> [preboot.log](https://github.com/user-attachments/files/16116529/preboot.log)

You shouldn't use EmuVariable. It will not remember entries.