CloverHackyColor/CloverBootloader #703: Compatibiliy issue with nvme and p35 chipset

ovlx, 2024-06-30:
Self explanatory,
The drive (kingston nv2 500gb ssd) is accessible  by os,
the pcie device can be seen with the efishell pci command , and is labeled as a "Mass Storage Controller - Non-volatile memory subsystem". but does not appear as a blk device or file system
The nvme driver is loaded  successfully, but no volume is detected  according to the debug log
The device  is never accessed by clover,  I can confirm this because adapter I have has an access light,
 the device  is treated like a non-storage device.

 this is occurring  on a p35 lga775 board , tested with a netburst cpu (no ssse3)  and a 45nm quad (with sse4.1) , and on 2 different p35 motherboards, and the result is the same,  

using the same clover usb and nvme drive on a modern am4 motherboard, the drive is bootable in both bios(csm) and uefi mode.

I'm using the latest build of clover 5159,  I did try an older build 5142 and had the same problem.

I can only speculate, but i'm assuming the issue is with the nvme driver? maybe it doesn't like pcie 1.0, I don't have a newer gen2 capable motherboard unfortunately, so i can't test that.
[2024-06-30_18-54_CLOVERX64.EFI.log](https://github.com/user-attachments/files/16046316/2024-06-30_18-54_CLOVERX64.EFI.log)

p5-vbnekit, 2024-09-18:
It works for me!!!

```
Colver Release: 5160.
CPU: Intel Core 2 Quad Q9650.
Motherboard: ASUS P5Q Premium.
MBR bootloader: `boot7`
NVMe storage: Samsung NVMe M.2 SSD 980 PRO, 1TB.
```

I hope this will help developers find and fix bugs.

It works if you:
1. Select `Efi Shell` in `Clover` menu.
2. In efi shell you can find loaded driver `nvmexpressdxe.efi` via command `drivers` but no one device attached (type `connect` without arguments to check it).
3. Type command `reconnect -r` and you will see new attached driver to your `nvme` device.
4. Type command `map -r` and your `nvme` partitions should be mapped.
5. Next type `map` without arguments and you will see your nvme `blkX:` partitions (`fs0:` with nvme `efi` `gpt` partition in my case).
6. Next you can chainload some next `efi` bootloader: `fs0:\EFI\Boot\bootx64.efi` for example - default removable `windows 11` bootloader in my case.

As temporary workaround i will try to create `startup.nsh` file to run my target os bootloader.

ovlx, 2025-01-03:
the p5q uses the p45 chipset, which supports pcie 2.0 which works fine from what I read.   I suspect there is some issue with detection on the older chipset.

Ultimately I was able to work around this issue with the p35 chipset by using inserting an NVME option rom into the bios, which allowed me to boot from it in legacy mode.

sambul13, 2025-12-03:
>**@p5-vbnekit** I will try to create `startup.nsh` file to run my target os bootloader.
  
You workaround works for me on Gigabyte GA-EP45-UD3P Mobo that is about the same 2018 or earlier year as your ASUS mobo. The only clarification to make is that after prepping the flash drive with BDU utility in Windows, open its volume and delete _boot_ entry in root directory, then rename _boot7_ to _boot_ to start it first as it shows Clover UI pretty fast.

Can you share your workaround script _startup.nsh_ and explain how to use or where to copy it on the flash drive? Or may be Clover developers can offer such script for now or integrate it into the latest ISO?