CloverHackyColor/CloverBootloader #704: Can't boot into Clover GUI when PCIe USB has an external HDD plugged in

eugenesvk, 2024-07-13:
After much trial and erorr I've managed to find a working version of clover Clover (old 5070) to boot Windows off an NVMe drive connected via the first PCIe→NVMe card for some rather old PC (as well as maintaining bootability via Clover into the old SSD)! Thanks!

But then I've discovered a strange bug: if I have an old HDD MBR NTFS plugged into that second PCIe→USB3 card, clover fails to boot into its GUI, it gets stuck at the `ScanVolumes` stage after successully enumerating all the drives.

Do you know why this non-bootable drive blocks Clover GUI?
Is there some other config I've missed that would allow ignoring all of that (I've tried with `Scan` parameters disabled, the attached config with my latest try has them enabled; also tried differnt 0-3 values for `LegacyBiosDefaultEntry`, but this shouldn't matter since this affects booting after Clover is loaded?)?

Interestingly enough, when I attach another external HDD via the same port (but this time GPT partitioned one that also has a Windows system partition), the GUI loads fine

The config and debug logs
- [config](https://gist.github.com/eugenesvk/608e255c341c438fe3c5489272c6d6c5#file-config-plist)
- [failed debug](https://gist.github.com/eugenesvk/608e255c341c438fe3c5489272c6d6c5#file-debugfail-exthdd-usb-c-pcie) the only difference is that it just stops after enumerating all the drives, otherwise identical
- [successful debug](https://gist.github.com/eugenesvk/608e255c341c438fe3c5489272c6d6c5#file-debug-ok-no-external-hdd)

<details>
<summary>Some system details</summary>

- Clover: v5070 burned USB Flash drive (MBR FAT32 via BDUtility) with the following adjustments:
  + add `GrubNTFS.efi` drivers to recognize Windows bootloader for the original Sata SSD MBR NTFS (so I can't remove it not to lose the ability to "dual"-boot into the old Windows)
  + add `NvmExpressDxe.efi` drivers to recognize NVMe drives
  + add `EmuVariableUefi` to set NVRAM and autoboot after a timeout
- BIOS is set to boot from HDD first and USB with Clover is manually selected first from the list (BIOS doesn't see any of these PCIe drives, neither the NVMe nor the USB3 ones)
- MB : BIOS/legacy only (no UEFI option) Gigabyte GA-EX58-UD5 (2008) [lnk](https://www.gigabyte.com/Motherboard/GA-EX58-UD5-rev-10/sp#sp)
- CPU : Intel Core i7-920 (2008) [lnk](https://www.intel.com/content/www/us/en/products/sku/37147/intel-core-i7920-processor-8m-cache-2-66-ghz-4-80-gts-intel-qpi/specifications.html)
- NVMe M.2 SSD: Samsung 980 500G (2001), GPT UEFI 3 partition (EFI system, Windows, data)
- old SATA SSD, MBR 2 partitions (system and recovery)
- external HDD that blocks Clover when connected via PCIe USB3: also old MBR with 2 paritions: NTFS and exFAT, not bootable, no secret EFI system partition
- external HDD that works with Clover with the same connection: GPT drive with a few paritions, including a system Windows one, so has EFI System Partition
- PCIe card for **NVMe**
- PCIe card for **USB3**

</details>

SergeySlice, 2024-09-10:
When someone report a possible bug he should use latest Clover version because all bugs already resolved.

closed 2024-09-10 by SergeySlice

eugenesvk, 2024-09-10:
That's not true because the new versions don't work at all, so its supposedly resolved bugs are even worse