CloverHackyColor/CloverBootloader #707: Help! How to uninstall Clover_r5159.pkg?

andy2301, 2024-08-09:
Friends, I accidentally installed this package Clover_r5159.pkg to my Macbook Air (while tinkering to install clover to a usb drive), and I cannot find a way to uninstall it. Please help. Thank you. 

Tried all the EFI-mounting approaches mentioned in #68 but could not find EFI partition on my Macbook (MacOS Monterey 12.5). 

`andy@MacAir /Applications % sudo diskutil list
/dev/disk0 (internal):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                         500.3 GB   disk0
   1:             Apple_APFS_ISC                         524.3 MB   disk0s1
   2:                 Apple_APFS Container disk3         494.4 GB   disk0s2
   3:        Apple_APFS_Recovery                         5.4 GB     disk0s3

/dev/disk3 (synthesized):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      APFS Container Scheme -                      +494.4 GB   disk3
                                 Physical Store disk0s2
   1:                APFS Volume Macintosh HD            23.4 GB    disk3s1
   2:              APFS Snapshot com.apple.os.update-... 23.4 GB    disk3s1s1
   3:                APFS Volume Preboot                 777.9 MB   disk3s2
   4:                APFS Volume Recovery                1.6 GB     disk3s3
   5:                APFS Volume Macintosh HD - Data     356.8 GB   disk3s5
   6:                APFS Volume VM                      2.1 GB     disk3s6`

A quick system-wide search showed these files were added around the same time when I installed the Clover_r5159.pkg.

`andy@MacAir / % sudo find / -iname '*clover*'
/private/var/db/receipts/org.clover.drivers64.mandatory.englishdxe.bom
/private/var/db/receipts/org.clover.drivers64.mandatory.englishdxe.plist
/private/var/db/receipts/org.clover.post.bom
/private/var/db/receipts/org.clover.themes.purple.swirl.plist
/private/var/db/receipts/org.clover.post.plist
/private/var/db/receipts/org.clover.pre.bom
/private/var/db/receipts/org.clover.utils.bom
/private/var/db/receipts/org.clover.cloverefi.64.sata.plist
/private/var/db/receipts/org.clover.efifolder.bom
/private/var/db/receipts/org.clover.themes.cesium.plist
/private/var/db/receipts/org.clover.themes.purple.swirl.bom
/private/var/db/receipts/org.clover.themes.clovy.bom
/private/var/db/receipts/org.clover.drivers64uefi.apfsdriverloader.uefi.bom
/private/var/db/receipts/org.clover.themes.bgm.plist
/private/var/db/receipts/org.clover.drivers64.mandatory.xhcidxe.plist
/private/var/db/receipts/org.clover.themes.clovy.plist
/private/var/db/receipts/org.clover.drivers64.mandatory.xhcidxe.bom
/private/var/db/receipts/org.clover.themes.cesium.bom
/private/var/db/receipts/org.clover.drivers64uefi.mandatory.englishdxe.uefi.bom
/private/var/db/receipts/org.clover.drivers64.apfsdriverloader.plist
/private/var/db/receipts/org.clover.efifolder.plist
/private/var/db/receipts/org.clover.biosboot.bom
/private/var/db/receipts/org.clover.drivers64uefi.fat.uefi.bom
/private/var/db/receipts/org.clover.biosboot.plist
/private/var/db/receipts/org.clover.drivers64uefi.mandatory.englishdxe.uefi.plist
/private/var/db/receipts/org.clover.drivers64uefi.apfsdriverloader.uefi.plist
/private/var/db/receipts/org.clover.utils.plist
/private/var/db/receipts/org.clover.bootloader.bootno.bom
/private/var/db/receipts/org.clover.bootloader.bootno.plist
/private/var/db/receipts/org.clover.cloverefi.64.sata.bom
/private/var/db/receipts/org.clover.themes.bgm.bom
/private/var/db/receipts/org.clover.drivers64uefi.openruntime.uefi.plist
/private/var/db/receipts/org.clover.pre.plist
/private/var/db/receipts/org.clover.drivers64uefi.fat.uefi.plist
/private/var/db/receipts/org.clover.drivers64.apfsdriverloader.bom
/private/var/db/receipts/org.clover.drivers64uefi.openruntime.uefi.bom`

maxpicelli, 2024-08-12:
Maybe you need try this program 

https://github.com/chris1111/Drop-EFI/releases/tag/V3 

Its made by @chris1111  that he is a developer team from Clover

andy2301, 2024-08-16:
Thanks, @maxpicelli. I tried Chris's latest release: installed it, opened it, dragged the "Macintosh HD" into it, and it said "Error: cannot locate EFI partition". 

BTW, I also checked the disk partitions from "diskutil list" and the files inside. To my eyes, none of the partitions appears to be an EFI partition.

Is there anything else I can check to uninstall the Clover_r5159.pkg?

maxpicelli, 2024-08-16:
I've never come across such a situation, but if I were you, I would try either the recovery tool, or if you can switch to the mode where you can boot from a pendrive, disabling some things on the Mac, I would use GParted from Linux instead. pendrive and delete everything. Very simple to use and very easy to format whatever you need. Then create a GUID Partition with GParted using the disk just for that purpose. After deleting it, if I were you I would restore it to the factory, but if you can, restore it from an ISO compatible with your use. 

https://gparted.org/download.php

Good Lucky my friend

maxpicelli, 2024-08-16:
BTW, today @chris1111  made a new release from Drop-EFI  3 hours ago. Try it too. 
 
https://github.com/chris1111/Drop-EFI/releases/

closed 2024-09-10 by SergeySlice