CloverHackyColor/CloverBootloader #708: shimex64.efi

Leubeling, 2024-08-28:
Bootloader Clover does not use/find shimx64.efi when secureboot is enabled.
- Clover boots with/without SecureBoot
- Windows boots with/without SecureBoot
- Ubuntu24.04 boots without SecureBoot (grubx64.efi)
- Ubuntu24.04 does not boot with SecureBoot (shimx64.efi)

Instead of selecting shimx64.efi, which loads grubx64.efi from the same path, there is a hardcoded string in the file `rEFIt_UEFI/entry_scan/loader.cpp::1472`
`XStringW File = SWPrintf("EFI\\%ls\\grubx64.efi", DirEntry->FileName);` which will create a entry only for grubx64.efi.

Unfortunately, due to several errors under ubuntu24.04, I am also unable to record a successful build. Therefore I cannot test my assumption and submit it as a pullrequest.

SergeySlice, 2024-09-05:
I can add such string into the sources but I can't test if it will work.

Leubeling, 2024-09-05:
hello SergeySlice,
... testing is no problem if I have an image with this fix to recreate my USB stick. 

I would test if 
- grubx64 would boot without secureboot
- and with shimx64

Also I would test if
- Boot/Secure has an influence (true or false)
- Boot/DefaultLoader has an influence (grubx64.efi or shimx64.efi)

SergeySlice, 2024-09-05:
Test please. Hope success
[BOOTX64.efi.zip](https://github.com/user-attachments/files/16899053/BOOTX64.efi.zip)

If OK then I will commit. If no then provide me some more information. preboot.log (by F2)  is obligatory.

SergeySlice, 2024-09-05:
Don't hurry, I can wait.

Leubeling, 2024-09-11:
So, first of all, thank you.

For me personally, I can already say that the goal has been achieved. I can start all my systems under UEFI-SecureBoot as well as UEFI-normally.

Of course, I also looked at your commit, which is why the test result is not surprising. It always seems to find grubx64.efi and shimx64.efi. Regardless of any settings.

**Testing (uefi SecureBoot):**
- windows -> found -> starts -> ok
- grubx64.efi -> found -> does not start as expected -> acceptable because you can hide it
- shimx64.efi -> found -> starts -> ok
- The “Secure” setting has no influence.

**Testing (uefi without SecureBoot):**
- windows -> found -> starts -> ok
- grubx64.efi -> found -> starts -> ok
- shimx64.efi -> found -> starts -> ok
- The “Secure” setting has no influence.

**Conclusion:**
I cannot say exactly whether all systems work with shimx64.efi or some also directly with grubx64.efi. That's why I would leave it as it is. If you don't want to see grubx64, you can hide it. Because of me it may be transferred to the release, thank you.

SergeySlice, 2024-09-12:
Thank you for the testing!

closed 2024-09-12 by SergeySlice