CloverHackyColor/CloverBootloader #712: BCM4352/BCM20702A0 bluetooth not working with Clover on Sequoia

cornelutz17, 2024-09-26:
Hi there,

im encounter some problems with bluetooth not working with Clover on Sequoia an i dont know how to fix this.
The weird thing is that if i boot with Opencore i still have the same problem, but if i boot with Opencore and i restart my laptop on the booting process and booting after that with Clover bluetooth is fully fuctional.

I donkt know if this has to do with the kext order or with some boot-args that cant be added to Clover.

Does someone had the same problem or know a fix for this?

Thnaks

SergeySlice, 2024-10-14:
AFAIK you should set in macOS
`sudo nvram bluetoothInternalControllerInfo=%00`
and reboot

cornelutz17, 2024-10-16:
done that, doesnt work

SergeySlice, 2024-12-17:
Choose recent Clover version.

cornelutz17, 2024-12-27:
I'm on r5161 and still encounter the same problem. Are there some quirks or options that i need to check/uncheck?

thanks

closed 2024-12-27 by cornelutz17

AntonOpri4nina, 2026-05-02:
> problem solved
> 
> thanks

Hello, I apologise for writing quite late, that is, after 2 years, but I faced the same problem. As a result, wherever I look for a solution, the topic is either closed due to the fact that the author has found a solution. Either because the topic is silent because they write standard non-working solutions to change kexts places or prescribe useless boot-args. If you don't mind, please remember how you solved this problem and write here. Thank you in advance!