CloverHackyColor/CloverBootloader #714: Clover cannot boot Ubuntu 24.04 (legacy bios)

hydra1one, 2024-11-11:
Hello.I installed ubuntu 24.04. Now clover can't boot it. Ubuntu 22.04 booted fine. I manually added the uuid of the ubuntu partition to UEFI/ubuntu/grub.cfg. But the system just goes into a reboot. The system is installed on another disk.If you boot via f8, it loads normally.Can you say something about this?I understand that something has changed in the ubuntu download.But I don't know what to do about it.Thank you.

ja26gu, 2024-11-25:
You don't even need clover for that!

SergeySlice, 2024-12-17:
I also don't know what is happen with Ubuntu 24.04.
But what did you do? You say legacy BIOS?
1. BIOS search a bootable drive and started here "boot" file which is Clover. OK? It doesn't depend on UBUNTU version.
2. This legacy part of Clover search and start CloverX64.EFI which shows you graphical boot menu. OK? You did see it?
3. Then you choose linux icon and press on it. And what is happen with Ubuntu 22.04 or with 24.04? Any differences?

hydra1one, 2024-12-19:
> I also don't know what is happen with Ubuntu 24.04. But what did you do? You say legacy BIOS?
>
>     1. BIOS search a bootable drive and started here "boot" file which is Clover. OK? It doesn't depend [...]

День добрый.Извиняюсь.Проще для меня будет объяснить на русском.Так как я знаю,что вы на нём говорите.
Суть проблемы.Стояла на отдельном диске и грузилась через меню кловера ubuntu 22.04.Mac os стоит на отдельном диске и там же установлен кловер.Обновил ubuntu до 24.04 путём полной переустановки .И теперь,если нажать на знчёк линукса в меню кловера, то вместо загрузки системы,выкидывает в консоль груба.(не в меню груба) Если вручную добавить id раздела с убунтой в 
EFI/ubuntu/grub.cfg в консоль уже не выкидывает а на пару секунд появляетя тёмносерый экран и система уходит в ребут.
Такое у меня впервые.Железо старое,Обычный биос.Если грузиться напрямую с диска с убунтой,то все нормально грузится.Спасибо.

SergeySlice, 2025-07-14:
К сожалению, не могу сказать ничего по сути вопроса, у меня нет средств воспроизвести ситуацию.
Могу рекомендовать ArchLinux
https://wiki.archlinux.org/title/Clover