CloverHackyColor/CloverBootloader #715: R730 Clover 5160 SSD drives not showing in the main menu only displayed when F3 is pressed

spdouts, 2024-11-24:
Hi, I require some assistance please where I want the above mentioned drive to automatically appear in the main menu instead of having to press F3 to display it.

spdouts, 2024-12-02:
Hello

Anyone suggestions please?

SergeySlice, 2024-12-17:
May be your config.plist contains some "Hide" instruction?

spdouts, 2024-12-17:
Thank you for your response

Below is the default config.plist

Can you please assist me with which sections I would need to update to show ESXI 7.0.3 in the main menu?

</array>
	<key>GUI</key>
	<dict>
		<key>ConsoleMode</key>
		<string>0</string>
		<key>Custom?</key>
		<dict>
			<key>Entries</key>
			<array>
				<dict>
					<key>Disabled</key>
					<false/>
					<key>Hidden</key>
					<false/>
					<key>Image</key>
					<string>os_mageia</string>
					<key>Path</key>
					<string>\EFI\MAGEIA\GRUBX64.efi</string>
					<key>Title</key>
					<string>Mageia Linux</string>
					<key>Type</key>
					<string>Linux</string>
				</dict>
				<dict>
					<key>AddArguments</key>
					<string>-v</string>
					<key>Arguments</key>
					<string>Kernel=mach_kernel</string>
					<key>Disabled</key>
					<true/>
					<key>Hidden</key>
					<false/>
					<key>Hotkey</key>
					<string>M</string>
					<key>InjectKexts</key>
					<false/>
					<key>NoCaches</key>
					<false/>
					<key>Path</key>
					<string>\EFI\BOOT\BOOTX64.efi</string>
					<key>Title</key>
					<string>MyCustomEntry</string>
					<key>Type</key>
					<string>OSXRecovery</string>
					<key>Volume</key>
					<string>D68F1885-571C-4441-8DD5-F14803EFEF54</string>
				</dict>
				<dict>
					<key>Hidden</key>
					<false/>
					<key>InjectKexts</key>
					<false/>
					<key>NoCaches</key>
					<false/>
					<key>SubEntries</key>
					<array>
						<dict>
							<key>AddArguments</key>
							<string>-v</string>
							<key>Title</key>
							<string>Boot OS X 10.8.5 (12F36) Mountain Lion in Verbose Mode</string>
						</dict>
					</array>
					<key>Title</key>
					<string>OS X 10.8.5 (12F36) Mountain Lion</string>
					<key>Type</key>
					<string>OSX</string>
					<key>Volume</key>
					<string>454794AC-760D-46E8-8F77-D6EB23D2FD32</string>
				</dict>
			</array>
			<key>Legacy</key>
			<array>

GlennHD, 2025-03-15:
I followed these instructions to just boot to the UUID of the device I wanted clover to boot to. The UUID can be found in the clover menu by hitting space bar on the selected hidden device. This was on an R720 where I was booting from PCIe NVMe drives. I setup my R730 6mo ago and vaguely remember doing the same thing.

https://forum.proxmox.com/threads/bootable-nvme-install-on-old-hardware-made-easy-with-pcie-adapter-and-clover.78120/

SergeySlice, 2025-04-06:
So the issue was resolved?

GlennHD, 2025-04-06:
I'm not OP but the documentation on the workarounds are plenty for these non-standard setups.