CloverHackyColor/CloverBootloader #723: No video output - KVM

Niaxor, 2025-01-05:
I have a KVM switch (UGREEN DP 1.4 Switch) which gets no video output when booting into Clover. Clover is still operational, and I can blindly select my OS and boot into it. 

Once booted into an OS, the OS (Arch, Windows etc..) has no issues displaying through the KVM. 

If I by-pass the KVM (Dp straight to monitor from GPU) there are no issues in Clover and menu is displayed as normal. 

Looking for any tips to debug further or understand root cause of the issue. The KVM is very useful to me :/

SergeySlice, 2025-01-17:
Will BIOS outputs to the KVM?

Niaxor, 2025-02-04:
No, it does not :(

Niaxor, 2025-02-04:
I'm wondering if its due to the BIOS not outputting video because the added startup lag to identify my monitor due to KVM.

BIOS is being lazy and not outputting video if there is no monitor detected at a specific point in boot cycle.

SergeySlice, 2025-04-06:
> No, it does not :(

Clover can output only if BIOS is capable to do this.

closed 2025-04-06 by SergeySlice