CloverHackyColor/CloverBootloader #726: Can you please create Clover more user friendly with better documentation and for Linux too?

Slavomir9, 2025-01-17:
Dear developers,
can you please create Clover more user friendly with better documentation and for Linux too?
I found any guide for Linux for Clover 6 years old. It is probably not actual and guide is too difficult. 
Can you please create any video guide?

I am not programmer. For me is use Clover too difficult.

Thank you

SergeySlice, 2025-01-17:
As far as I know you should not make any effort to boot Linus from Clover. It just works.

juanpc2018, 2025-02-08:
Try [DUET 2020](https://winraid.level1techs.com/t/guide-nvme-boot-for-systems-with-legacy-bios-and-uefi-board-duet-refind/32251)
or 
rEFInd

Grub is the Linux default bootloader.

Slavomir9, 2025-02-09:
[SergeySlice](https://github.com/SergeySlice)
Any guides for Linux is possible found 
[Clover EFI Bootloader Manual Installation via Ubuntu (UEFI)  ](https://www.youtube.com/watch?v=YPWWinxwOcY)
 [Boot Ubuntu and Linux Mint ISO from Clover Bootloader](https://www.youtube.com/watch?v=ODa9ZW4HGEw)

Slavomir9, 2025-02-09:
[juanpc2018](https://github.com/juanpc2018)
Thank you very much for answer. 
Yes. I use rEFInd. It is simple.
Advantage of Clover is more active development.

SergeySlice, 2025-04-06:
May be you should read a manual
https://github.com/CloverHackyColor/CloverBootloader/releases/download/5160/Clover_Of_Khaki_Color_5160-en.pdf

SergeySlice, 2025-04-06:
> Try [DUET 2020](https://winraid.level1techs.com/t/guide-nvme-boot-for-systems-with-legacy-bios-and-uefi-board-duet-refind/32251) or rEFInd
> 
> Grub is the Linux default bootloader.

It looks like advertising  of extraneous work.

Slavomir9, 2025-04-07:
[SergeySlice](https://github.com/SergeySlice)
Thank you very much for guide

juanpc2018, 2025-06-16:
Problem 1.
DUET 2020 does Not work with Dual CPU server Bios boards.

My Goal is similar to DUET 2020
Boot Legacy BIOS USB MBR pendrive -> NVME drivers ->
UEFI GPT Linux & UEFI MBR Windows 8.1x64 on BIOS machine.

# test if Clover also has the same problem of DUET 2020 in Dual CPU server Bios boards. <--

Problem 2.
installing Clover for Legacy MBR on Linux or Windows

using Rufus / Unetbootin to burn Clover.iso to USB as MBR
 does Not work like LinuxLiveCD .iso

Clover recomends GPT ssd/hdd, 
but Linux Grub goes crazy, if has GPT SSD + Bios Grub & Machine.
Linux Grub requires GPT for UEFI, and MBR for Legacy / Bios Boot.
maybe some distros work, but Ubuntu 20.04 LTS does Not.

Linux workaround:
 copy Linux partition from GPT ssd to MBR ssd, 
using Gparted of the same version LiveCD.iso
then use something like Boot-Rescue.iso to create the MBR Bios Grub,
or install a New Linux on the same SSD, on a different partition.

Windows 8.1x64
BCDBootloader can be manualy forced to be Bios & UEFI = bcdboot.exe /f ALL
i have tested SSD MBR /f ALL
works ok in DUET 2020.

Reading the manual page 31 onwards: