CloverHackyColor/CloverBootloader #730: [FAIL] Asmedia ASM 1062 PCIe to SATA-III 6Gbps

juanpc2018, 2025-02-07:
https://www.sonnettech.com/product/tempossd.html
http://ableconn.com/support_1.php?gid=120

Latest Firmware [220916_00_00_00 A23](http://ableconn.com/upload/ASM1062-FW_220916-RE1062.A23.zip)

Asmedia are "Generic AHCI"
based on a modified i8051 core.
as far as i know.

Direct to SATA-2 3Gbps on the board works ok.

SergeySlice, 2025-04-01:
You mean legacy boot from an SSD attached to this controller? Are you sure the controller has a legacy mode?
I know the same fail with VLC controller. There is documented absence of legacy capability.

juanpc2018, 2025-04-01:
i've done a lot of tests, im trying to remember.

all i remember is that ASMedia AS1062 AHCI mode does Not work with USB Bootloaders,
Bootloaders AHCI drivers does Not detect AS1062 generic AHCI controller.

i've tested old boards without UEFI,
New boards with UEFI.
PC & Mac

Win8.1x64 requires BIOS mode FAT32 partition to boot on Apple intel machines, like 2010 Mac Pro5,1 or 2014 mac mini.
does Not detect Win8.1 SSD, IF  FAT32 partition has BCDboot /f ALL "Bios+EFI", or /f EFI 
Only works /f BIOS on intel Apple machines.

DUET 2020 Bootloader "based on Clover+rEFInd"
detects Win8.1x64 with /f ALL "Bios+EFI"
msinfo32.exe detects both modes: UEFI  & BIOS ok.
connected directly to Board SATA-II 3Gbps.

also detects Linux 20.04.4 LTS GPT+EFI Grub2

DUET 2020 does Not Boot IF board has 2x CPUs, Only boot with 1x CPU.
also has other problems with too Memory test.

Newer boards like ASRock Z790 LiveMixer
require Real GPU: GTX 1050 Ti or similar to boot Legacy-CSM mode,
intel integrated GPU legacy support was removed in 12th gen CPUs,
10th gen work ok,
11th gen unknown.

board also requires UEFI v9.03, 
latest UEFI v18 does Not work well with 12th gen CPUs, 
example: Thunderbolt4 PCIe card does Not work on Linux, but works ok with UEFI v9.03
UEFI under <v9.03 does Not work with 14th Gen CPUs.

intel Macs OSX 10.6.8 - OSX Catalina 10.15.7, requires EFI, GPU and EFI partition,
OSX Catalina is 100% intel, 100% 64-Bit,
OSX HighSierra is the last 32 & 64-Bit.

ASMedia AS1062 required Firmware Update to solve some compatibility issues in OSX.
example: Apple allows to move the SSD to different machines, booting from USB3.0 or internal SATA,
IF machines are compatible with OSX version,
but older AS1062 Firmware did Not allowed to move the SSD, only worked if was installed on that machine.
that was solved in 200xx Firmware.

Windows is Fixed to a single machine, Pro version allows to move SSD, but requires phone Re-Activation.
Windows OEM is Node/Machine Locked, can only be activated in 1 board, Never moved.

Linux is portable.

IF board works EFI & Bios boot modes using a USB Bootloader,
but ASMedia 1062 controller does Not allow.

------------------------

Sonnet version
```
Mac Compatibility

Mac Pro® 3,1, 4,1 (2009), 5,1 (2010-2012), 7,1 (2019), or 14,8 (2023)
Power Mac® G5 (with PCIe slots)
Xserve® (Intel with PCIe slots)
OS X® 10.5+
macOS Sonoma compatible

PC Compatibility
PC with PCIe slots
Windows® 11, 10, 8, 7
Windows Server 2019, 2016, 2012, 2008 R2, 2008

Thunderbolt Compatibility
[... 155 lines cut ...]
ATAPI Support
Yes
Physical Characteristics
Weight of Product
2.1 oz [59.0 g]
Product Height
0.7 in [1.8 cm]
Product Length
4.7 in [12.0 cm]
Product Width
2.1 in [5.4 cm]
Software
OS Compatibility
Windows® 7, 8, 8.1, 10, 11
Windows Server® 2012 R2, 2016, 2019, 2022
macOS® 10.10 to 15
Linux 2.6.32 and up - LTS Versions only
What's in the Box
Included in Package
1 - PCIe SATA Card
1 - Full-Profile Bracket (Installed)
1 - Low-Profile Bracket
1 - Quick-Start Guide

```

SergeySlice, 2025-04-06:
You forgot to mention that a SATA controller itself should allow legacy (BIOS) mode to operate.