CloverHackyColor/CloverBootloader #735: What is the architecture of CloverBootloader?

ohault, 2025-03-16:
Would it possible to document CloverBootloader architecture, internal design and how it works?

SergeySlice, 2025-04-01:
I am not sure if I understand your question.
Architecture is x86_64. There are some old codes for i386 but unsupported and there are some templates for ARM but not in a development.
Internal design? How it works? Read the manual
https://github.com/CloverHackyColor/CloverBootloader/releases/download/5160/Clover_Of_Khaki_Color_5160-en.pdf

ohault, 2025-04-01:
> I am not sure if I understand your question.
> Architecture is x86_64. There are some old codes for i386 but unsupported and there are some templates for ARM but not in a development.
> Internal design? [...]

Thank you @SergeySlice 

Indeed, I was not referring to neither the supported computer architecture nor the supported Instruction Set Architecture, but the internal high level design of CloverBootLoader.

Can you consider to publish the manual in html too?

SergeySlice, 2025-04-06:
Is it enough?
https://github.com/CloverHackyColor/CloverBootloader/wiki
Anyway PDF is also readable by a browser.

SergeySlice, 2025-05-07:
I am not sure if I understand your question.
Architecture is x86_64. There are some old codes for i386 but unsupported and there are some templates for ARM but not in a development.
Internal design? How it works? Read the manual